nanopb/nanopb #379: `oneof` must have at least one field

sureshjoshi, 2019-02-12:
Using the `master` branch of nanopb, I have not been able to get any .proto files with `oneof` working. My other proto files, without `oneof` work correctly.

I'm using 3.6.1 protoc installed via Homebrew (3.6.1.3)

Proto2: sample.proto

```
syntax = "proto2";

message Bar {
  optional bytes data = 1;
}

message Foo {
  oneof value {
    Bar bar = 1;
  }
}
```

```
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1861, in <module>
    main_plugin()
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1846, in main_plugin
    results = process_file(filename, fdesc, options, other_files)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1727, in process_file
    headerdata = ''.join(f.generate_header(includes, headerbasename, options))
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1404, in generate_header
    yield msg.fields_declaration(self.dependencies) + '\n'
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1028, in fields_declaration
    defval = self.default_value(dependencies)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1127, in default_value
    desc = google.protobuf.descriptor.MakeDescriptor(optional_only)
  File "/usr/local/Cellar/protobuf/3.6.1.3_1/libexec/lib/python2.7/site-packages/google/protobuf/descriptor.py", line 1015, in MakeDescriptor
    _message.default_pool.Add(file_descriptor_proto)
TypeError: Couldn't build proto file into descriptor pool!
Invalid proto descriptor for file "d68bf30050157c4439e43ac1aba6144e.proto":
  Foo.value: Oneof must have at least one field.

--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

Proto3: sample.proto

```
syntax = "proto3";

message Foo {
  oneof value {
    int32 bar = 1;
  }
}
```

```
Traceback (most recent call last):
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1861, in <module>
    main_plugin()
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1846, in main_plugin
    results = process_file(filename, fdesc, options, other_files)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1727, in process_file
    headerdata = ''.join(f.generate_header(includes, headerbasename, options))
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1404, in generate_header
    yield msg.fields_declaration(self.dependencies) + '\n'
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1028, in fields_declaration
    defval = self.default_value(dependencies)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1127, in default_value
    desc = google.protobuf.descriptor.MakeDescriptor(optional_only)
  File "/usr/local/Cellar/protobuf/3.6.1.3_1/libexec/lib/python2.7/site-packages/google/protobuf/descriptor.py", line 1015, in MakeDescriptor
    _message.default_pool.Add(file_descriptor_proto)
TypeError: Couldn't build proto file into descriptor pool!
Invalid proto descriptor for file "5df690209b200d56fb41c82ebf9a1f91.proto":
  Foo.value: Oneof must have at least one field.

--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2019-02-12:
I've pushed a possible fix (in two commits) for this problem, but for some reason I'm having trouble reproducing it myself so can't be sure if it fixes it.

If problem still occurs, could you try adding in nanopb_generator.py line 1129 before `MakeDescriptor()` call a print statement and post the debug info:

        sys.stderr.write("DESCRIPTOR: %s\n" % repr(optional_only))

sureshjoshi, 2019-02-12:
Thanks for the quick reply!

As a result of the latest master branch, same proto3 as above:

```
syntax = "proto3";

message Foo {
  oneof value {
    int32 bar = 1;
  }
}
```

I get (with the added sys.stderr):

```
DESCRIPTOR: name: "Foo"
oneof_decl {
  name: "value"
}

  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1864, in <module>
    main_plugin()
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1849, in main_plugin
    results = process_file(filename, fdesc, options, other_files)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1730, in process_file
    headerdata = ''.join(f.generate_header(includes, headerbasename, options))
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1407, in generate_header
    yield msg.fields_declaration(self.dependencies) + '\n'
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1028, in fields_declaration
    defval = self.default_value(dependencies)
  File "/Users/sureshjoshi/Developer/Scratch/nanopb/generator/nanopb_generator.py", line 1129, in default_value
    optional_only.ClearField('oneof_decl')
TypeError: field name must be a string
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

If I remove the commit that added 'ClearField' - then we go back to the original problem

Unfortunately, I don't understand what is going on here, because I'd like to pitch in to help any way that I can.

PetteriAimonen, 2019-02-12:
Thanks, that at least gives more information. Can you check output of this command:

    python -c 'from google.protobuf.internal import api_implementation; print(api_implementation._default_implementation_type)'

Based on https://github.com/protocolbuffers/protobuf/issues/3658 it seems the reason you're having different results could be if your system has `cpp` implementation, while mine has `python` implementation of Google's library. I didn't even know there are two different implementations, I'll have to add this to my test systems.

I pushed a further fix that might now work.

sureshjoshi, 2019-02-12:
```
python -c 'from google.protobuf.internal import api_implementation; print(api_implementation._default_implementation_type)'
cpp
```

And yes, that latest fix works. Thanks!

PetteriAimonen, 2019-02-12:
I'll keep this open so that I'll remember to add this to the test suite.

closed 2019-12-20 by PetteriAimonen