nanopb/nanopb #380: mangle_names:M_STRIP_PACKAGE is broken when message name (partially) matches package name

paullieverse-w, 2019-02-18:
When trying to use nanopb 0.3.9.2 with `mangle_names:M_STRIP_PACKAGE`, we encountered an issue. It looks like this feature is broken in case a message name (partially) matches the package name.

For example, the following results in code that can be compiled with `mangle_names:M_STRIP_PACKAGE`, and thus seems to work correctly:
```
syntax = "proto2";
package P;
message A {
    message B {
        optional uint32 val = 1;
    }
    optional B b = 1;
}
```

But when the package name matches the message name, code gets generated (again, with `mangle_names:M_STRIP_PACKAGE`) that cannot be compiled anymore:
```
syntax = "proto2";
package A;
message A {
    message B {
        optional uint32 val = 1;
    }
    optional B b = 1;
}
```
Which results in a compiler error:
```
In file included from test_mangle_names2.pb.c:4:0:
test_mangle_names2.pb.h:20:5: error: unknown type name ‘B’
     B b;
     ^
In file included from test_mangle_names2.pb.h:6:0,
                 from test_mangle_names2.pb.c:4:
test_mangle_names2.pb.c:14:66: error: ‘B_fields’ undeclared here (not in a function); did you mean ‘A_fields’?
     PB_FIELD(  1, MESSAGE , OPTIONAL, STATIC  , FIRST, A, b, b, &B_fields),
                                                                  ^
../../nanopb-0.3.9.2-linux-x86/pb.h:418:30: note: in definition of macro ‘PB_OPTIONAL_STATIC’
     pb_membersize(st, m), 0, ptr}
                              ^~~
test_mangle_names2.pb.c:14:5: note: in expansion of macro ‘PB_FIELD’
     PB_FIELD(  1, MESSAGE , OPTIONAL, STATIC  , FIRST, A, b, b, &B_fields),
     ^~~~~~~~
```
Examining the generated source code and comparing to that of the previous example, the struct generated for the `B` is called `A_B`, whereas the member inside `A` has type `B` (and not `A_B`). Same for the referenced `B_fields` (there is a definition of `A_B_fields`).

And when the message name starts with the package name, the generator itself (once again with `mangle_names:M_STRIP_PACKAGE`) even fails:
```
syntax = "proto2";
package P;
message PA {
    message B {
        optional uint32 val = 1;
    }
    optional B m = 1;
}
```
```
Traceback (most recent call last):
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1789, in <module>
    main_plugin()
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1774, in main_plugin
    results = process_file(filename, fdesc, options, other_files)
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1651, in process_file
    f = parse_file(filename, fdesc, options)
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1633, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1126, in __init__
    self.parse()
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 1184, in parse
    self.messages.append(Message(name, message, message_options))
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 930, in __init__
    field = Field(self.name, f, field_options)
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 405, in __init__
    self.ctype = self.submsgname = names_from_type_name(desc.type_name)
  File "../nanopb-0.3.9.2-linux-x86/generator/nanopb_generator.py", line 130, in names_from_type_name
    raise NotImplementedError("Lookup of non-absolute type names is not supported")
NotImplementedError: Lookup of non-absolute type names is not supported
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2019-02-18:
@matejcik Are you interested in looking into this problem?

matejcik, 2019-02-21:
i'll try to take a look sometime next week

PetteriAimonen, 2019-03-07:
I'm looking into this now, trying to get a fix into 0.3.9.3. Looks like for some reason the bug only occurs when using `protoc --nanopb_out=...`, when doing `protoc -ofoo.pb foo.proto` and `python nanopb_generator.py foo.pb` it seems to work correctly.

PetteriAimonen, 2019-03-07:
Should be fixed now, in both master and 0.3 maintenance branch.

Bug was caused by how the plugin mode processes each file twice to check for dependencies. `mangle_field_typename()` modifies the structure, and it ended up stripping the prefix twice if it appeared to still contain the prefix. Fixed by taking a copy of the message structure before each processing pass.

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen