nanopb/nanopb #381: Unable to compile proto file that uses variable length data type

GnUfTw, 2019-02-25:
**Background**
* Nanopb version: 0.3.9.2
* Platform: Windows

**Issue**
I have downloaded the latest version of nanopb and unzipped it to a location in my home directory. I have placed the .proto file I want to compile at the root of the unzipped source directory. The .proto file I am trying to compile is shown below (slimmed down) and is called MyMessages.proto:
```
syntax = "proto3";
import "generator/proto/nanopb.proto";
...
message MyMessage {
    repeated sint32 my_message = 1 [(nanopb).max_count = 500];
}
...
```
The command I am using to compile the .proto file is shown below:
```
.\generator-bin\protoc.exe -o MyMessages.pb .\MyMessages.proto --csharp_out=".\" --proto_path=".\"
```
Running this command fails with the following output:
```
google/protobuf/descriptor.proto: File not found.
generator/proto/nanopb.proto: Import "google/protobuf/descriptor.proto" was not found or had errors.
generator/proto/nanopb.proto:108:8: "google.protobuf.FileOptions" is not defined.
generator/proto/nanopb.proto:112:8: "google.protobuf.MessageOptions" is not defined.
generator/proto/nanopb.proto:116:8: "google.protobuf.EnumOptions" is not defined.
generator/proto/nanopb.proto:120:8: "google.protobuf.FieldOptions" is not defined.
MyMessages.proto: Import "generator/proto/nanopb.proto" was not found or had errors.
```
I have confirmed that nanopb.proto & descriptor.proto both exist in the expected locations. Any idea what I might be doing wrong here?

PetteriAimonen, 2019-02-25:
Try adding `--proto-path .\generator\proto`

GnUfTw, 2019-02-25:
@PetteriAimonen You da man! Thanks for the suggestion. 

I modified my protobuf compilation command to the following and I am now able to successfully generate my nanopb source files that use a variable length data type:
```
.\generator-bin\protoc.exe -o MyMessages.pb .\MyMessages.proto --csharp_out=".\" --proto_path=".\" --proto_path=".\generator\proto"
```

closed 2019-02-25 by GnUfTw