nanopb/nanopb #382: Am I encoding a message that uses a fixed size float array correctly?

GnUfTw, 2019-02-25:
I have using a fixed-size repeated field in my .proto file as follows:
```
syntax = "proto3";

message DeliveryComplete {
    repeated float delivered_data = 1;
}

message ControllerToHostMessageContainer {
    oneof msg {
        .....
        DeliveryComplete delivery_complete = 3;
    }
}
```
I have a complementary .options file defined as follows:
```
DeliveryComplete.delivered_data     max_count:512   fixed_count:true
```
In order to use an array of this size I have to pass in a switch to my compiler as follows:
```
-D PB_FIELD_16BIT
```
I have a function that populates the message container, encodes it, wraps the encoded message in a frame and writes the data. It is implemented as follows:
``` c
// Relevant declarations.
#define APP_MAKE_BUFFER_DMA_READY  __attribute__((coherent)) __attribute__((aligned(16)))
#define APP_WRITE_BUFFER_SIZE (1024)
#define DELIVERED_HISTORY_BUFFER_SIZE (512)
static uint8_t APP_MAKE_BUFFER_DMA_READY encodedBuffer[APP_WRITE_BUFFER_SIZE];
static uint8_t APP_MAKE_BUFFER_DMA_READY generatedBuffer[APP_WRITE_BUFFER_SIZE];
static int encodedBufferLength;
static int generatedBufferLength;

void SendHostData(const float* deliveryData)
{
    // Populate message container
    ControllerToHostMessageContainer container = (ControllerToHostMessageContainer) ControllerToHostMessageContainer_init_zero;
    int deliveryDataIndex;
    for (deliveryDataIndex = 0; deliveryDataIndex < DELIVERED_HISTORY_BUFFER_SIZE; deliveryDataIndex++)
    {
        container.msg.delivery_complete.delivered_data[deliveryDataIndex] = *deliveryData;
        deliveryData++;
    }
    container.which_msg = ControllerToHostMessageContainer_delivery_complete_tag;

    // Encode protobuf message.
    memset(encodedBuffer, 0, APP_WRITE_BUFFER_SIZE);
    pb_ostream_t stream = pb_ostream_from_buffer((pb_byte_t *) encodedBuffer, (size_t) APP_WRITE_BUFFER_SIZE);
    pb_encode(&stream, ControllerToHostMessageContainer_fields, &container);
    encodedBufferLength = stream.bytes_written;
    
    // Wrap encoded message in frame.
    generatedBufferLength = MESSAGE_FRAMING_Generate(encodedBuffer, encodedBufferLength, generatedBuffer, APP_WRITE_BUFFER_SIZE);

    // Logic to write data to host goes here.
    USB_DEVICE_CDC_Write(appData.cdcInstance, &appData.writeTransferHandle, generatedBuffer, generatedBufferLength, USB_DEVICE_CDC_TRANSFER_FLAGS_DATA_COMPLETE);
}
```
If I examine the contents of the **container** I create in the function above, using my debugger (after the for loop) it appears to me that the message container is being populated with all of the values I would expect to be there (specifically around 50 non-zero floating point values). But, if I examine the value of **encodedBufferLength** after the statement `encodedBufferLength = stream.bytes_written;` it has a value of 3 which doesn't seem correct to me, it should be significantly higher.

I am using the nanopb API correctly to encode this message or am I missing something?

GnUfTw, 2019-02-25:
I have convinced myself that I am using the API correctly, closing the issue is related to something I am doing wrong outside of nanopb land.

GnUfTw, 2019-02-25:
Maybe I spoke too soon, it would be good to hear it from someone who knows for sure. Re-opening.

PetteriAimonen, 2019-02-26:
You may be hitting this issue with proto3 syntax and fixed length messages: https://github.com/nanopb/nanopb/issues/376
It would occur if the first item in your array has value zero.

Try this fix: https://github.com/nanopb/nanopb/commit/6c772817cafb6f45f446fef70746a89c8dfdfd47

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen