nanopb/nanopb #386: Potential bug in encoding of proto3 repeated callback fields

polytypic, 2019-03-07:
We ran into a potential bug in encoding of proto3 repeated callback fields in 0.3.9.1 (also in 0.3.9.2).  Here is roughly the proto file we are implementing:

```protobuf
syntax = "proto3";

message Value {
    Primitive primitive = 1;
    Structure structure = 2;
    Array array = 3;
}

enum Type {
    NONE = 0;
    BOOL = 1;
    INT8 = 2;
    UINT8 = 3;
    INT16 = 4;
    UINT16 = 5;
    INT32 = 6;
    UINT32 = 7;
    INT64 = 8;
    UINT64 = 9;
    FLOAT = 10;
    DOUBLE = 11;
    STRING = 12;
}

message Primitive {
    Type type = 1;
    bytes data = 2;
}

message Property {
    string name = 1;
    Value value = 2;
}

message Structure {
    repeated Property properties = 1;
}

message Array {
    repeated Value values = 1;  // <--- The problem?
}
```

The above basically specifies a kind of JSON like recursive data structure.

When trying to encode an `Array`, Nanopb attempts [at line 228 of `pb_encode.c`](https://github.com/nanopb/nanopb/blob/nanopb-0.3.9.2/pb_encode.c#L228) to read the size of the array.  However, this does not make sense, because there is callback field rather than a field with size:

```c
/* Enum definitions */
typedef enum _Type {
    Type_NONE = 0,
    Type_BOOL = 1,
    Type_INT8 = 2,
    Type_UINT8 = 3,
    Type_INT16 = 4,
    Type_UINT16 = 5,
    Type_INT32 = 6,
    Type_UINT32 = 7,
    Type_INT64 = 8,
    Type_UINT64 = 9,
    Type_FLOAT = 10,
    Type_DOUBLE = 11,
    Type_STRING = 12
} Type;
#define _Type_MIN Type_NONE
#define _Type_MAX Type_STRING
#define _Type_ARRAYSIZE ((Type)(Type_STRING+1))

/* Struct definitions */
typedef struct _Array {
    pb_callback_t values;
/* @@protoc_insertion_point(struct:Array) */
} Array;

typedef struct _Structure {
    pb_callback_t properties;
/* @@protoc_insertion_point(struct:Structure) */
} Structure;

typedef struct _Primitive {
    Type type;
    pb_callback_t data;
/* @@protoc_insertion_point(struct:Primitive) */
} Primitive;

typedef struct _Value {
    Primitive primitive;
    Structure structure;
    Array array;
/* @@protoc_insertion_point(struct:Value) */
} Value;

typedef struct _Property {
    pb_callback_t name;
    Value value;
/* @@protoc_insertion_point(struct:Property) */
} Property;
```

```c
const pb_field_t Value_fields[4] = {
    PB_FIELD(  1, MESSAGE , SINGULAR, STATIC  , FIRST, Value, primitive, primitive, &Primitive_fields),
    PB_FIELD(  2, MESSAGE , SINGULAR, STATIC  , OTHER, Value, structure, primitive, &Structure_fields),
    PB_FIELD(  3, MESSAGE , SINGULAR, STATIC  , OTHER, Value, array, structure, &Array_fields),
    PB_LAST_FIELD
};

const pb_field_t Primitive_fields[3] = {
    PB_FIELD(  1, UENUM   , SINGULAR, STATIC  , FIRST, Primitive, type, type, 0),
    PB_FIELD(  2, BYTES   , SINGULAR, CALLBACK, OTHER, Primitive, data, type, 0),
    PB_LAST_FIELD
};

const pb_field_t Property_fields[3] = {
    PB_FIELD(  1, STRING  , SINGULAR, CALLBACK, FIRST, Property, name, name, 0),
    PB_FIELD(  2, MESSAGE , SINGULAR, STATIC  , OTHER, Property, value, name, &Value_fields),
    PB_LAST_FIELD
};

const pb_field_t Structure_fields[2] = {
    PB_FIELD(  1, MESSAGE , REPEATED, CALLBACK, FIRST, Structure, properties, properties, &Property_fields),
    PB_LAST_FIELD
};

const pb_field_t Array_fields[2] = {
    PB_FIELD(  1, MESSAGE , REPEATED, CALLBACK, FIRST, Array, values, values, &Value_fields),
    PB_LAST_FIELD
};
```

We came up with two different potential fixes for this.  Initially we tried:

```diff
         /* Required proto2 fields inside proto3 submessage, pretty rare case */
         return false;
     }
-    else if (PB_HTYPE(type) == PB_HTYPE_REPEATED)
+    else if (PB_HTYPE(type) == PB_HTYPE_REPEATED && field->size_offset)
     {
         /* Repeated fields inside proto3 submessage: present if count != 0 */
         return *(const pb_size_t*)pSize == 0;
```

and then we tried:

```diff
```diff
         /* Required proto2 fields inside proto3 submessage, pretty rare case */
         return false;
     }
-    else if (PB_HTYPE(type) == PB_HTYPE_REPEATED)
+    else if (PB_HTYPE(type) == PB_HTYPE_REPEATED && PB_ATYPE(type) != PB_ATYPE_CALLBACK)
     {
         /* Repeated fields inside proto3 submessage: present if count != 0 */
         return *(const pb_size_t*)pSize == 0;
```

Both of the above seem to fix the problem (encoding and decoding then work as expected) and also do not seem to break any existing Nanopb tests.

Does this seem like a bug in Nanopb 0.3.9.1 (and 0.3.9.2) and is one or the other of the above attempts a correct fix?

PetteriAimonen, 2019-03-07:
Yeah, looks like #376 also affects callbacks. The fix I added is in commit 6c772817cafb6f4, which will consider fixed length arrays and callback fields as "always present".

The git master has had a larger cleanup done to `pb_check_proto3_default_value()` which also added a check for `PB_ATYPE_STATIC`.

I guess it's time to release 0.3.9.3, as this bug is getting a lot of hits lately.

polytypic, 2019-03-07:
Thanks for the quick reply!

Any estimate on when 0.3.9.3 release could be available?

PetteriAimonen, 2019-03-07:
Maybe next week. I'll have to go through changes and see if there is anything else that should be backported.

The code itself is available in maintenance_0.3 branch

PetteriAimonen, 2019-03-08:
Fix released in nanopb-0.3.9.3.

closed 2019-03-08 by PetteriAimonen