nanopb/nanopb #387: Can't import message definition into a new message in 0.3.9.3

toodv93, 2019-03-20:
Made a simple upgrade from 0.3.9.2 to 0.3.9.3, but getting errors when generating file b.proto.

File: a.proto:
```
syntax = "proto3";

message Msg_A
{
  uint32 msg_a = 1;
}
```

File b.proto:
```
syntax = "proto3";

import "a.proto";

message Msg_B
{
  Msg_A test = 1;
}
```

Now in 0.3.9.3 I get following:

> Traceback (most recent call last):
>   File "generator/nanopb_generator.py", line 1890, in <module>
>     main_cli()
>   File "generator/nanopb_generator.py", line 1791, in main_cli
>     results = process_file(filename, None, options)
>   File "generator/nanopb_generator.py", line 1755, in process_file
>     sourcedata = ''.join(f.generate_source(headerbasename, options))
>   File "generator/nanopb_generator.py", line 1501, in generate_source
>     yield msg.fields_definition(self.dependencies) + '\n\n'
>   File "generator/nanopb_generator.py", line 1057, in fields_definition
>     width = self.required_descriptor_width(dependencies)
>   File "generator/nanopb_generator.py", line 1073, in required_descriptor_width
>     max_offset = self.data_size(dependencies)
>   File "generator/nanopb_generator.py", line 1093, in data_size
>     return sum(f.data_size(dependencies) for f in self.fields)
>   File "generator/nanopb_generator.py", line 1093, in <genexpr>
>     return sum(f.data_size(dependencies) for f in self.fields)
>   File "generator/nanopb_generator.py", line 597, in data_size
>     raise Exception("Unhandled field type: %s" % self.pbtype)
> Exception: Unhandled field type: MESSAGE

(using libprotoc 3.0.0)

I just unpacked nanopb-0.3.9.3.tar.gz and tried to start to use it. Should i have done something more? Maybe I have missed something.

PetteriAimonen, 2019-03-21:
Hi,

> I just unpacked nanopb-0.3.9.3.tar.gz and tried to start to use it. 
> Should i have done something more? Maybe I have missed something.

Uh oh, I've made a mistake. I've accidentally created the 
nanopb-0.3.9.3.tar.gz tar from the master branch instead of the 
maintenance branch, and you are getting hit by bug #377.

You can confirm this by checking NANOPB_VERSION in pb.h, if it says 
nanopb-0.4.0-dev, you have the erroneous package.

I've now updated the package on the download page.

--
Petteri

toodv93, 2019-03-21:
Thanks!
Now it works fine.
Issue can be closed

closed 2019-03-21 by toodv93