nanopb/nanopb #390: Makefile:4: recipe for target 'nanopb_pb2.py' failed. Make: *** [nanopb_pb2.py] Error 127

rogersguedes, 2019-04-23:
At commit `72f4594`, running `make` inside the `generator/proto` fails if the `protoc` is not in `PATH` and there is no check for env var `PROTOC`:
```
$ cd generator/proto
$ make
protoc --python_out=. nanopb.proto
make: protoc: Command not found
Makefile:4: recipe for target 'nanopb_pb2.py' failed
make: *** [nanopb_pb2.py] Error 127
```
Applying the patch bellow should fix:
```
diff --git a/generator/proto/Makefile b/generator/proto/Makefile
index 12db23f..c17fb9c 100644
--- a/generator/proto/Makefile
+++ b/generator/proto/Makefile
@@ -1,4 +1,6 @@
+PROTOC?=protoc
+
 all: nanopb_pb2.py
 
 %_pb2.py: %.proto
-	protoc --python_out=. $<
+	$(PROTOC) --python_out=. $<
```

rogersguedes, 2019-04-23:
Thank you @PetteriAimonen, your contributions are awesome!

PetteriAimonen, 2019-12-20:
Fix published now in 0.4.0.

Also note that there is now also __init__.py that calls protoc automatically, so using the makefile is not strictly required.

closed 2019-12-20 by PetteriAimonen