nanopb/nanopb #392: Error while generating code for submessages described in other file.

Uraty, 2019-04-24:
I'm using master brunch, because I'm interested in callbacks for oneOf https://github.com/nanopb/nanopb/issues/175

I have created two files. `inbound.proto`:

```
syntax = "proto3";
package device.messaging.device;

import "device/messaging/device/v1/inbound.proto";

message DeviceInboundMessage {
    oneof version {
        v1.InboundV1 v1 = 1;
    }
}
```

And `v1/inbound.proto`

```
syntax = "proto3";
package device.messaging.device.v1;

message InboundV1 {
    oneof type {
        ...
    }
}
```

As result I have next error:

```
Traceback (most recent call last):
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1893, in <module>
    main_cli()
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1794, in main_cli
    results = process_file(filename, None, options)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1758, in process_file
    sourcedata = ''.join(f.generate_source(headerbasename, options))
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1504, in generate_source
    yield msg.fields_definition(self.dependencies) + '\n\n'
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1060, in fields_definition
    width = self.required_descriptor_width(dependencies)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1076, in required_descriptor_width
    max_offset = self.data_size(dependencies)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1096, in data_size
    return sum(f.data_size(dependencies) for f in self.fields)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 1096, in <genexpr>
    return sum(f.data_size(dependencies) for f in self.fields)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 836, in data_size
    return max(f.data_size(dependencies) for f in self.fields)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 836, in <genexpr>
    return max(f.data_size(dependencies) for f in self.fields)
  File "/Users/dmitry/Downloads/nanopb-gh/generator/nanopb_generator.py", line 600, in data_size
    raise Exception("Unhandled field type: %s" % self.pbtype)
Exception: Unhandled field type: MESSAGE
```

I have looked into `nanopb_generator.py` with debugger. So I've seen `dependencies` argument, that doesn't contain target message type (it contains only messages, described in same file).

Posible workarounds:

1. Define messages in the same file.
2. For messages that only encoded - `[(nanopb).type = FT_POINTER]` for fields which types described in separated file. (maybe it can work for decoding as well, but I can't understand how to work with such pointers during decoding)
3. Use callback (`repeated`, or `[(nanopb).type = FT_CALLBACK]`). But it doesn't work with oneOf.

PetteriAimonen, 2019-04-24:
It might help if you run `nanopb_generator.py` in the top folder of which `device/messaging/device/v1/` is a subdir, or specify `-I` with the top directory as an argument. Then nanopb will find the included file and can read information about the submessage from there.

But this issue is still a bug, because it should resort back to safe defaults if the submessage cannot be read.

Uraty, 2019-04-24:
Thank you!
It works, but important detail (if I've correctly understood): it must be run as plugin for `protoc`.
In `nanopb_generator.py`, `main_cli()` call has `process_file(filename, None, options)` - without `other_files` argument, so it uses default value `{}`. `other_files` is filled within `main_plugin` call.

PetteriAimonen, 2019-12-20:
Actually fixed in commit 5e5280de13, forgot to update issue report. Fix is now published in 0.4.0.

closed 2019-12-20 by PetteriAimonen