nanopb/nanopb #393: Undefined symbols for architecture

fthuin, 2019-04-25:
Hello,
I'm getting an error while building my project (iOS application) with Xcode 10.2.1.

```
Ld /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Products/Debug-iphoneos/nanopb/nanopb.framework/nanopb normal arm64 (in target: nanopb)
    export IPHONEOS_DEPLOYMENT_TARGET=8.0
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang -arch arm64 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS12.2.sdk -L/Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Products/Debug-iphoneos/nanopb -F/Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Products/Debug-iphoneos/nanopb -filelist /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/nanopb.build/Objects-normal/arm64/nanopb.LinkFileList -install_name @rpath/nanopb.framework/nanopb -Xlinker -rpath -Xlinker @executable_path/Frameworks -Xlinker -rpath -Xlinker @loader_path/Frameworks -miphoneos-version-min=8.0 -dead_strip -Xlinker -object_path_lto -Xlinker /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/nanopb.build/Objects-normal/arm64/nanopb_lto.o -Xlinker -export_dynamic -Xlinker -no_deduplicate -fobjc-arc -fobjc-link-runtime -framework Foundation -compatibility_version 1 -current_version 1 -Xlinker -dependency_info -Xlinker /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Intermediates.noindex/Pods.build/Debug-iphoneos/nanopb.build/Objects-normal/arm64/nanopb_dependency_info.dat -o /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Products/Debug-iphoneos/nanopb/nanopb.framework/nanopb
```
```
Undefined symbols for architecture arm64:
  "_pb_field_iter_begin", referenced from:
      _pb_decode_noinit in pb_decode.o
      _pb_message_set_to_defaults in pb_decode.o
      _pb_release in pb_decode.o
      _iter_from_extension in pb_decode.o
      _pb_encode in pb_encode.o
      _pb_check_proto3_default_value in pb_encode.o
  "_pb_field_iter_find", referenced from:
      _pb_decode_noinit in pb_decode.o
      _pb_release_union_field in pb_decode.o
  "_pb_field_iter_next", referenced from:
      _pb_decode_noinit in pb_decode.o
      _find_extension_field in pb_decode.o
      _pb_message_set_to_defaults in pb_decode.o
      _pb_release in pb_decode.o
      _pb_encode in pb_encode.o
      _pb_check_proto3_default_value in pb_encode.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

And if I go to ` /Users/fthuin/Library/Developer/Xcode/DerivedData/myproject-ezsfqzsknxmnkphdnsnekleddfjl/Build/Products/Debug-iphoneos/nanopb`, then the structure is

```
sirius:nanopb fthuin$ tree
.
└── nanopb.framework
    ├── Headers
    │   ├── nanopb-umbrella.h
    │   ├── pb.h
    │   ├── pb_common.h
    │   ├── pb_decode.h
    │   └── pb_encode.h
    └── Modules
        └── module.modulemap
```

So the path in the first command line might not exist but I don't know why because it looks everything is autogenerated.

PetteriAimonen, 2019-04-25:
It looks like `pb_common.c` is not included in your project file.

fthuin, 2019-04-25:
@PetteriAimonen You are right, `pb_common.c` is not included in my Pods, could it mean that the latest Pod release is missing a file?

PetteriAimonen, 2019-04-25:
I don't know either, I don't use Firebase or pods and I don't know who is maintaining them.

trcarden, 2019-07-11:
@PetteriAimonen : This is is craaaaazy! nanopb is a dependency for the Firebase CORE library. Like every single person trying to use Flutter + Firebase on iOS (if I am right) will hit this right now.

I bet $5 bucks that the only reason why few people have gotten this far is because Flutter is like 10 abstraction levels above this issue and people trying flutter don't know how any of the machinery works.

In any case let me see if I can figure out how to get the right people involved to fix it

## Apendix

For reference and maybe SEO

1. Here is the Google docs where most normal people will start their journey: https://firebase.google.com/docs/analytics/ios/start

[image]

2. Even if you don't install the Analytics package you will get it via the FirebaseBase ... Core Pod usually https://github.com/CocoaPods/Specs/blob/master/Specs/0/3/5/Firebase/6.4.0/Firebase.podspec.json

[image]

3. This brings us to the Firebase Analytics Pod which somehow is a dependency of Core (bad Google) is defined here: https://github.com/CocoaPods/Specs/blob/master/Specs/e/2/1/FirebaseAnalytics/6.0.3/FirebaseAnalytics.podspec.json.

[image]

4. Finally that lands us here https://cocoapods.org/pods/nanopb and the Maintainer `Jiayong Ou`

[image]

5. However after it turns out that is a lie and the real maintainer at Google is... @rsgowman or @paulb777

https://github.com/google/nanopb-podspec 
and his (likely) broken pubspec file:

https://github.com/google/nanopb-podspec/blob/master/nanopb/0.3.9.1/nanopb.podspec

(missing the file in question `pb_common.c` in the sources)

[image]

I will file a ticket against that pubspec to see if they can investigate

trcarden, 2019-07-11:
@PetteriAimonen My mistake, it was simply a bad cache and with the cocoa pods system.

If anyone else runs into this you can simply:

1. `rm -rf Pods`
2. `pod install --verbose`
3.  Get something like this
```
> Copying nanopb from `[redacted]Library/Caches/CocoaPods/Pods/Release/nanopb/0.3.901-2901f` to `Pods/nanopb`
  - Running pre install hooks
```
4. Simply destroy that bad cache `rm [redacted]Library/Caches/CocoaPods/Pods/Release/nanopb/0.3.901-2901f`
5. Then destroy your pods and rerun `pod install --verbose` to get a fresh pull
[image]

According to the pubspec those files should be in there.

closed 2019-12-19 by PetteriAimonen

pk359, 2021-04-08:
> @PetteriAimonen My mistake, it was simply a bad cache and with the cocoa pods system.
> 
> If anyone else runs into this you can simply:
> 
> 1. `rm -rf Pods`
> 2. `pod install --verbose`
> 3. Get something like this
> 
> ```
> > Copying nanopb from `[redacted]Library/Caches/CocoaPods/Pods/Release/nanopb/0.3.901-2901f` to `Pods/nanopb`
>   - Running pre install hooks
> ```
> 
> 1. Simply destroy that bad cache `rm [redacted]Library/Caches/CocoaPods/Pods/Release/nanopb/0.3.901-2901f`
> 2. Then destroy your pods and rerun `pod install --verbose` to get a fresh pull
>    [image]
> 
> According to the pubspec those files should be in there and the linker will find it and everything should work.

Yes works.