nanopb/nanopb #394: mangle_names M_FLATTEN too aggressive and/or a new option ?

smorin-primero, 2019-04-28:
I've been running into a problem, I have multiple proto files with some dependencies.
Each proto file have their own subpackage and the package name are getting too big for my compiler.

I've tried generating using `M_FLATTEN`, but I ran into problems immediately, something like:

```
message Foo{
  enum Result{
    OK = 1
  }
 Result result = 1;
}

message Bar{
  enum Result{
    YES = 1
  }
 Result result = 1;
}
```
Would end up with both Result stripped to the same name and conflicting.

M_STRIP_PACKAGE doesn't work either because I have some message with the same name in both `Foo.Bar` and `Foo.Qux`.

From the description, `M_FLATTEN` seems to be doing it's job, but I'm not sure it should be this thorough.

Otherwise, maybe something like `M_PACKAGE_INITIALS` could be used, so instead of having the full package name I could get only the initials and that way I would be able to cut the variable names a lot and end up with something still organized by package.

Would that be something worth looking into ?

PetteriAimonen, 2019-04-28:
Yeah, sounds reasonable. I won't have time to implement this any time soon, but if you want to take a look yourself, the implementation should be relatively similar to M_STRIP_PACKAGE.

smorin-primero, 2019-04-28:
Great, I will look into it tomorrow

smorin-primero, 2019-04-30:
@PetteriAimonen I`ve been trying to make some of the changes needed, and I keep running into problems. 

I've set myself to try and debug the nanopb_generator.py directly and I see that there is support for a CLI mode, which I expect is there to help with debugging, but I can't seem to pass the right parameters to the command line to make parsing work.

I was trying to use the proto files generated by the typename_mangling test, any pointer you could give me for that ?

PetteriAimonen, 2019-04-30:
This should work:

<pre>
petteri@oddish:~/nanopb/tests/typename_mangling$ protoc -owith_package.pb with_package.proto 
petteri@oddish:~/nanopb/tests/typename_mangling$ python ../../generator/nanopb_generator.py with_package.pb 
Following patterns in ./with_package.options did not match any fields: TopLevelMessage.ShortIfStripPackage, ShortIfFlatten
Use  protoc --nanopb-out=-v:.   to see a list of the field names.
Writing to with_package.pb.h and with_package.pb.c
</pre>

The warnings about unmatched patterns are just because the .options file has the variants for all stripping options.

smorin-primero, 2019-04-30:
Oh, thank you, here I was trying to load the proto file directly, this makes much more sense.

closed 2019-12-20 by PetteriAimonen