nanopb/nanopb #395: encode results hex 1a 00, after updating nanopb from nanopb-0.3.9 to nanopb-0.3.9.3

farizadp, 2019-04-30:
Encoding result for the following message gives different result between those 2 versions.
```
syntax = "proto3";

message Error
{
    int32 code = 1;
    string message = 2;
}

message Reply
{
    enum Result
    {
        ERROR = 0;
        OK = 1;
    }

    Result result = 1;
    Error error = 2;
}
```
When I encode Reply.result = Reply_Result_OK 
nanopb-0.3.9 gives hex 0x08 0x01
nanopb-0.3.9.3 gives hex 0x08 0x01 0x1A 0x00

Have anyone experienced this? Why are those 1a00 at the end?

PetteriAimonen, 2019-04-30:
0x1A 0x00 is an empty submessage or string with tag 3.

But no idea yet why that would be produced. Can you post a larger snippet of the code you use to encode it? Perhaps there are some uninitialized fields in your structure?

farizadp, 2019-04-30:
Actually, I have more than 3 tags in my message definition. Here I only mention 3.

```
syntax = "proto3";

message ConfirmationRequest
{
    int64 id = 1;
    int32 timeout_sec = 2;
}

message Error
{
    int32 code = 1;
    string message = 2;
}

message Reply
{
    enum Result
    {
        ERROR = 0;
        OK = 1;
    }

    Result result = 1;
    Error error = 2;
    ConfirmationRequest confirmation_request = 3;
}
```

And this is how I encode the message
```
{
    pb_ostream_t ostream;
    uint8_t buffer[512] = {0};
    Reply reply = Reply_init_zero;
    bool ret = false;
    Reply_Result request_result = Reply_Result_OK;

    ostream = pb_ostream_from_buffer(buffer, sizeof(buffer));
    reply.result = request_result;
    if (!pb_encode(&ostream, Reply_fields, &reply)) {
        ret = false;
        goto exit;
    }

    LOGW("respond payload (%d):", ostream.bytes_written);
    for (uint32_t i = 0; i < ostream.bytes_written; i++) {
        printf("%02X", buffer[i]);
    }
    printf("\n");
    LOGI("payload (end)");

    return ret;
}
```

farizadp, 2019-04-30:
Tried switching back to 0.3.9 and the problem is gone. However, I'm curious if it's something that we coded wrong but v0.3.9 doesn't detect it. But on the other hand, I couldn't find any changes which is related to it in the changelog. Any clue?

PetteriAimonen, 2019-04-30:
It's likely related to bug #376 in some way. But for the next few weeks I'm quite busy so I don't have time to look into it much yet. It seems that the message should decode the same regardless, i.e. that it just results in two extra bytes that shouldn't matter.

farizadp, 2019-05-01:
Right, the thing is the receiving entity decodes this stream using ```protoc```. So they will see tag 3 and the empty string after decoding (and wonder why), while I actually intend to only send tag 1. For now, I keep using nanopb-0.3.9

PetteriAimonen, 2019-12-19:
Sorry for this taking so long, but I've now confirmed that the issue is reproducible and I'll fix it soon.

PetteriAimonen, 2019-12-19:
Fixed for 0.3 maintenance branch, fix will eventually land in 0.3.9.5.

In current master (0.4.0) branch, it had been fixed already as part of other clean up, but I added the regression test there also.

closed 2019-12-20 by PetteriAimonen