nanopb/nanopb #397: Problem compiling when using wrapper types

yatendra, 2019-04-30:
I am using google provided simple wrapper types that are just a message wrapper over primitive type. Following is how my proto file looks like - 

syntax = "proto3";
import "google/protobuf/wrappers.proto";
message MyMessage {
    oneof sendRecordThresholdField {
        uint64 sendRecordThreshold = 6 [deprecated = true]; // [Deprecated].
        google.protobuf.UInt64Value sendRecordThresholdWrapper = 11; 
    }
}

MyMessage.pb.h and MyMessage.pb.c files get generated fine but when I build a sample app using these files, it fails saying - 

In file included from MyMessage.pb.h:8,
                 from decode.c:9:
/usr/include/google/protobuf/wrappers.pb.h:7:10: fatal error: string: No such file or directory
 #include <string>
          ^~~~~~~~

FYI - it looks like there was a wrappers.proto file in /usr/include/google/protobuf and when the makefile executed protoc on MyMessage.proto file, it generated wrappers.pb.h for wrappers.proto in /usr/include/google/protobuf. 
Also do you think this problem has anything to do with "#include <string>" being a c++ syntax.
Thanks in advance.

PetteriAimonen, 2019-05-01:
That file is for Google's C++ protobuf library.

Try compiling `wrappers.proto` using nanopb generator and arranging the include paths so that it gets included instead of the C++ version. It's a bit annoying that both nanopb and the C++ library use the `.pb.h` extension so that files sometimes get confused for each other. If necessary, you can change that extension for nanopb by using command line option.

yatendra, 2019-05-02:
Thanks. It worked.

closed 2019-05-02 by yatendra