nanopb/nanopb #399: DeprecationWarning: 'U' mode is deprecated

ooxi, 2019-05-06:
When using nanopb I receive the following deprecation warning from [nanopb_generator.py](https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L1712)

```
DeprecationWarning: 'U' mode is deprecated
  Globals.separate_options = read_options_file(open(optfilename, "rU"))
```

How can we fix this warning? Unfortunately I'm not familiar enough with Python to fix it myself.

I'm using Arch Linux with Python 3.7.3

PetteriAimonen, 2019-05-07:
Hmm, it will need some check for Python 2 vs. Python 3. It is unfortunate that they've added that warning in Python 3, as it was easier to make code portable between the Python versions without it.

The fix would probably be to assign `openmode = 'rU'` vs. `openmode = 'r'` in the existing check on line 119.

For now you can ignore the warning.

ooxi, 2019-05-19:
Thanks for the feedback! I did not consider Python 2 ./. compatibility

closed 2020-02-02 by PetteriAimonen