nanopb/nanopb #402: Issues with proto.nano_pb2.M_PACKAGE_INITIALS

levvayner, 2019-05-24:
I have the following (on windows 10 machine)
D:\msys32\esp\nanopb\generator\proto>protoc -IVMC_Proto --plugin=protoc-gen-nanopb=.\..\protoc-gen-nanopb.bat --nanopb_out=. VMC_Proto\partial.proto
Traceback (most recent call last):
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1900, in <module>
    main_plugin()
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1880, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1726, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1244, in __init__
    self.parse()
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1261, in parse
    elif mangle_names == nanopb_pb2.M_PACKAGE_INITIALS:
AttributeError: module 'proto.nanopb_pb2' has no attribute 'M_PACKAGE_INITIALS'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

If I manually add that property to the proto.nanopb_pb2 file, I get the following error:
D:\msys32\esp\nanopb\generator\proto>protoc -IVMC_Proto --plugin=protoc-gen-nanopb=.\..\protoc-gen-nanopb.bat --nanopb_out=. VMC_Proto\partial.proto
Traceback (most recent call last):
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1900, in <module>
    main_plugin()
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1880, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1726, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1244, in __init__
    self.parse()
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 1314, in parse
    self.messages.append(Message(name, message, message_options))
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 881, in __init__
    self.load_fields(desc, message_options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 921, in load_fields
    field = Field(self.name, f, field_options)
  File "D:\msys32\esp\nanopb\generator\\nanopb_generator.py", line 314, in __init__
    self.callback_datatype = field_options.callback_datatype
AttributeError: callback_datatype
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

I've checked versions 
D:\msys32\esp\nanopb\generator\proto>protoc --version
libprotoc 3.8.0

D:\msys32\esp\nanopb\generator\proto>python -m pip install protobuf==v3.8.0rc1
Requirement already satisfied: protobuf==v3.8.0rc1 in c:\users\lev.vayner.vaynersystems\appdata\local\programs\python\python37\lib\site-packages (3.8.0rc1)
Requirement already satisfied: setuptools in c:\users\lev.vayner.vaynersystems\appdata\local\programs\python\python37\lib\site-packages (from protobuf==v3.8.0rc1) (40.8.0)
Requirement already satisfied: six>=1.9 in c:\users\lev.vayner.vaynersystems\appdata\local\programs\python\python37\lib\site-packages (from protobuf==v3.8.0rc1) (1.12.0)

I am trying to use the functionality you added in 0.4.0, so I cannot download one of the binary packages (worked fine when I did that with 0.9.3)

levvayner, 2019-05-24:
I've tried in on MSYS32. Downloaded ready image from ESP-IDF link, updated and installed all required tools and libraries with 
pacman -Syu
pacman -S base-devel gcc vim cmake
pacman -S development

downloaded protobuf 3.8.0rc1 repo and built it (no errors on build, error on one of 6 tests during make check).
cleaned out nanopb proto folder scripts and ran make to generate them again.

After all of that, same issue
$ protoc --plugin=protoc-gen-nanopb=protoc-gen-nanopb VMC_Proto/Model.proto --nanopb_out=.
Traceback (most recent call last):
  File "D:/msys32/esp/nanopb/generator/nanopb_generator.py", line 1900, in <module>
    main_plugin()
  File "D:/msys32/esp/nanopb/generator/nanopb_generator.py", line 1880, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "D:/msys32/esp/nanopb/generator/nanopb_generator.py", line 1726, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "D:/msys32/esp/nanopb/generator/nanopb_generator.py", line 1244, in __init__
    self.parse()
  File "D:/msys32/esp/nanopb/generator/nanopb_generator.py", line 1261, in parse
    elif mangle_names == nanopb_pb2.M_PACKAGE_INITIALS:
AttributeError: 'module' object has no attribute 'M_PACKAGE_INITIALS'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

PetteriAimonen, 2019-05-24:
`AttributeError: 'module' object has no attribute 'M_PACKAGE_INITIALS'`
This error sounds like you need to re-run `make` in `generator/proto` folder. It seems you have an older version of nanopb_pb2.py there. In theory the `__init__.py` script should re-generate that automatically, but maybe it's not finding your protoc.

If a binary package is preferred, I have development builds here:
http://jpa.kapsi.fi/jenkins/job/nanopb-binary-windows/

levvayner, 2019-05-24:
Thanks Patteri! that was super fast and binaries work much better for me, I think.
I grabbed your latest and tried with that. I think it has something to do with my model, because I can generate the google well known type proto classes just fine. 
Also it works on the older version..

C:\Users\lev.vayner.VAYNERSYSTEMS\Downloads\**nanopb-0.4.0-dev-74-ga9c207e-windows-x86**>generator-bin\protoc --nanopb_out=. generator-bin\vmc-proto\Model.proto
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1900, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1885, in main_plugin
  File "__main__protoc-gen-nanopb__.py", line 1766, in process_file
  File "__main__protoc-gen-nanopb__.py", line 1429, in generate_header
  File "__main__protoc-gen-nanopb__.py", line 1028, in fields_declaration
  File "__main__protoc-gen-nanopb__.py", line 1141, in default_value
  File "google\protobuf\reflection.py", line 114, in MakeClass
  File "google\protobuf\reflection.py", line 114, in MakeClass
  File "google\protobuf\reflection.py", line 119, in MakeClass
  File "google\protobuf\internal\python_message.py", line 162, in __init__
  File "google\protobuf\internal\python_message.py", line 288, in _AttachFieldHelpers
  File "google\protobuf\internal\python_message.py", line 265, in _IsMapField
AttributeError: 'NoneType' object has no attribute 'has_options'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

C:\Users\lev.vayner.VAYNERSYSTEMS\Downloads\nanopb-0.4.0-dev-74-ga9c207e-windows-x86>C:\Users\lev.vayner.VAYNERSYSTEMS\Downloads\**nanopb-0.3.9.3-windows-x86**\nanopb-0.3.9.3-windows-x86\generator-bin\protoc --nanopb_out=. generator-bin\vmc-proto\Model.proto

C:\Users\lev.vayner.VAYNERSYSTEMS\Downloads\nanopb-0.4.0-dev-74-ga9c207e-windows-x86>

Contents of the model file
======================================
syntax = "proto3";
option csharp_namespace = "xxx.Messaging.D2C";

import "WellKnownTypes/wrappers.proto";
import "WellKnownTypes/timestamp.proto";
import "WellKnownTypes/duration.proto";

// well known types from C:\Users\admin\.nuget\packages\google.protobuf.tools\3.4.0\tools\google\protobuf\

message DateTimeOffset {
	google.protobuf.Timestamp UtcTime = 1;
	google.protobuf.Duration Offset = 2;
}

message MessageBatch {
	repeated Message Messages = 1;
	}

message Message {

	uint32 Id = 1;
	DateTimeOffset Generated = 2;

	oneof concrete {

		HeartbeatMessage Heartbeat = 101;
		
		SoftwareErrorMessage SoftwareError = 201;
		
		PlanogramMessage Planogram = 301;
		
		TransactionMessage Transaction = 401;
		VendMessage Vend = 402;
		DeliveryCollectedMessage DeliveryCollected = 403;

		DoorStateMessage DoorState = 501;
		TemperatureMessage Temperature = 502;
		LowInventoryMessage LowInventory = 503;
		SelectionEmptiedMessage SelectionEmptied = 504;
		MachineConfigMessage MachineConfig = 505;
		
	}
	
	message HeartbeatMessage {

	}

	message SoftwareErrorMessage {

		enum ComponentEnum {
			Invalid = 0;
			Orchestrator=1;
			CloudCommunicator=2;
		}

		enum SeverityEnum {
			Invalid1 = 0;
			High=1;
			Medium=2;
			Low=3;
		}

		ComponentEnum Component = 1;
		SeverityEnum Severity = 2;
		string Description = 3;
		string Exception = 4;
	}

	message VendMessage {

		enum VendStateEnum {
			Invalid = 0;
			Started=1;
			Succeded=2;
			Failed=3;
		}

		uint32 ProductId = 1;
		VendStateEnum State = 2;
		google.protobuf.BytesValue SessionId = 3;
		google.protobuf.BytesValue TransactionId = 4;
	}

	message DeliveryCollectedMessage {

		google.protobuf.BytesValue SessionId = 1;
	}
	
	message TransactionMessage {
	
		enum TransactionStateEnum {
			Invalid = 0;
			Started=1;
			Succeded=2;
			Canceleld=3;
			Failed=4;
		}

		enum PaymentProviderEnum {
			Invalid2 = 0;
			Cash=1;
			CreditCard=2;
		}
		
		message ShoppingCartLineItem {

			uint32 ProductId = 1;
			uint32 Quantity = 2;
			float ItemPrice = 3;
		}

		bytes TransactionId = 1;
		TransactionStateEnum State = 2;
		DateTimeOffset StartedAt = 3;
		PaymentProviderEnum PaymentProvider = 4;
		float PaymentAmount = 5;
		float TotalPrice = 6;
		repeated ShoppingCartLineItem LineItems = 7;
		google.protobuf.BytesValue SessionId = 8;
	}
	
	message PlanogramMessage {
	
		message PlanogramProduct {

			uint32 MachineIndex=1;
			uint32 RowIndex=2;
			uint32 ColumnIndex=3;
			uint32 ProductId = 4;
			google.protobuf.FloatValue Price = 5;
			google.protobuf.UInt32Value Count = 6;
		}

		bool IsActive = 1;
		repeated PlanogramProduct ProductItems = 2;
		google.protobuf.BytesValue SessionId = 3;
	}
	
	message DoorStateMessage {
		bool IsOpen = 1;
	}

	message TemperatureMessage {
		float Sensor1 = 1;
		google.protobuf.FloatValue Sensor2 = 2;
	}
	
	message LowInventoryMessage {
		uint32 ProductId = 1;
		uint32 Quantity = 2;
		google.protobuf.BytesValue SessionId = 3;
	}

	message SelectionEmptiedMessage {
		uint32 MachineIndex=1;
		uint32 RowIndex=2;
		uint32 ColumnIndex=3;
		uint32 ProductId = 4;
		google.protobuf.BytesValue SessionId = 5;
	}
	
	message MachineConfigMessage {
	
		message Selection {

			uint32 MachineIndex=1;
			uint32 RowIndex=2;
			uint32 ColumnIndex=3;
		}

		repeated Selection Selections = 1;
		google.protobuf.BytesValue SessionId = 2;
	}
	
}

PetteriAimonen, 2019-05-24:
I've pushed a possible partial fix for this now, new binaries will be available on Jenkins soon.

It seems that the problem occurs somewhere in `default_value()` logic, which was the cause for issue #379 also. But I didn't have enough time to find the root cause yet.

However, in your case there are no default values anyway (because proto3 syntax doesn't have them), so it's unnecessary to even try to run that part of the code. So now it will skip that logic altogether for proto3 files. I'll study further later if there is more testing needed.

levvayner, 2019-05-24:
Thank you Patteri! I will look out for the binary in your build, since I have difficulty building the nanopb libraries on my MSYS32 and windows boxes (my limitations, nothing wrong with nanopb).
Please let me know if I can do something to help resolve this issue or test it, or if you need any additional data to reproduce/resolve.

PetteriAimonen, 2019-05-25:
Good to hear it worked :)

I'll keep this open as a reminder to myself because I want to study it a bit further when I have time.

PetteriAimonen, 2019-12-19:
Hopefully all the corner cases with default value generation have been fixed now. But it seems to be a bug-prone part of the generator code.

commit 37458f49 (2019-05-24): Skip default value generation when no fields have them

This might partially fix #402, but it still requires more testing.
Nevertheless, this fix makes sense because there is no point going
through all the reflection + serialization logic if there is nothing
to do anyway.
closed 2019-12-19 by PetteriAimonen