nanopb/nanopb #403: Inconsistent decoding results

Renha, 2019-05-31:
Hello. I have the code:
```
pb_ostream_t stream = pb_ostream_from_buffer(packet_data, packet_length);
if (pb_decode(&stream, message_fields, &parsed_data)){
  return 0;
} return 1;
```
which runs periodically. And while `packet_data` is always the same, `08 0D 20 FF 01 28 3C`,
the function returns sometimes 0, sometimes 1.

While trying to debug the program I've changed every `return false` at `pb_decode_noinit` to `return true`, and return status became stable. But when I've tried to change those one-by-one to find out which one fails, results were unstable, so more than one `return false` play dice.

Renha, 2019-05-31:
stream.errmsg is 0x00 every time

Renha, 2019-05-31:
Tried to make `stream` global variable. No luck.

Renha, 2019-05-31:
Oh, `pb_istream_from_buffer` should be used, my bad.

closed 2019-05-31 by Renha

PetteriAimonen, 2019-05-31:
Shouldn't that give a compilation warning? If you are trying to pass pb_ostream_t* to pb_decode()?