nanopb/nanopb #406: Nested oneof

BartHertog, 2019-06-06:
Hello,

Currently we are trying to encode a oneof nested in another oneof. Unfortunately this fails. Could you help me explain this behavior. We are not using callbacks.

We have a setup similar to the following example:

Envelop.proto
```proto
syntax = "proto3";

import "Config.proto";
import "State.proto";
import "Command.proto";

message Envelop
{
  uint64 time_stamp = 1;
  
  oneof data
  {
    Config  cfg   = 10;
    State   state = 11;
    Command cmd   = 12;
  }
}
```

For simplicity we now only list the Config.proto. The other messages are similar.
```proto
syntax = "proto3";

message ConfigA
{
  // Some data.
}

message ConfigB
{
  // Some data
}

message Config
{
  // Some other data
  
  oneof specific_config
  {
    ConfigA a = 6;
    ConfigB b = 7;
  }
}
```

Now we assign the messages
```C++
ConfigB b;
b.somedata = somevalue;

Envelop envelop = default_values;
envelop.which_data = config_tag;
envelop.cfg.which_specific_config = ConfigB_tag;
envelop.cfg.specific_config.b = b;

pb_ostream_t ostream = pb_ostream_from_buffer(data_ptr, ENVELOP_SIZE);
result = pb_encode(&ostream, envelop_fields, &envelop);
``` 

This fails to encode the specific_config enum. For some reason in pb_encode_submessage() when calling pb_encode() -> encode_field() -> encode_basic_field(), pSize is never equal to the tag. The result is that zero bytes are encoded of the nested Config message.

How can this be explained?

PetteriAimonen, 2019-06-06:
Can you set a breakpoint before the call to pb_encode(), and print out the whole `envelop` variable?

BartHertog, 2019-06-06:
Yes we did, forgot to mention that the structs are filled with valid data.

PetteriAimonen, 2019-06-06:
.. could you show the data?

BartHertog, 2019-06-06:
Sorry for that, I could have added that the first time:

```
envelop                               const envelop & {...}  
  timestamp_ms                        int64_t         21  
  which_content                       pb_size_t       10 '\n'  
  content  union                      {...}           {...}  
    board_config                      board_config    {...}  
      io_id                           uint32_t        0  
      last_config                     bool            false  
      which_specific_config           pb_size_t       7 '\a'  
      specific_config  union          {...}           {...}  
        discrete_output               dig_out_config  {...}  
          active                      activation      activation_On  
          current_measurement_active  bool            false  
          update_period_ms            uint32_t        100  
          inverted                    bool            10    
          samples                     uint32_t        0  
```
bytes_written = 4 after pb_encode.

PetteriAimonen, 2019-06-06:
I tried to reproduce the problem based on your description, but it seems to work correctly for me. The code in attached [issue_406.tar.gz](https://github.com/nanopb/nanopb/files/3263104/issue_406.tar.gz) produces 7 bytes `52 05 3a 03 10 d2 09` which decoded with [this tool](https://protogen.marcgravell.com/decode) gives two nested messages, as it should.

You could try debugging this by comparing the pSize pointer `encode_basic_field()` gets to the actual address of the `which_specific_config` field. If there is some difference in compilation settings between files, the struct layouts may differ and it gets wrong pointer and thus wrong data. Most common such problem have been mixed C/C++ setups where some C++ have different size for `bool` than the C compiler.

BartHertog, 2019-06-07:
The point you make about the layout is a good point. We worked on this today as we are indeed in a C and C++ environment. We changed out all booleans for an enum. Unfortunately this did not resolve the problem. What did resolve the problem was removing the oneof specific_config and placing ConfigA and ConfigB directly in Config. This worked as expected. 

We will continue on this late next week or early the week after.

PetteriAimonen, 2019-06-07:
One debugging step that can help is make both a C file and C++ file and print `sizeof(MyMessage)` from both. If you get different values, there is a mismatch between the compiler settings.

As this is something that usually affect other code than nanopb also (such as function calling between C and C++ code), it is best to fix it in your compiler settings.

closed 2019-12-16 by PetteriAimonen