nanopb/nanopb #407: Extension fields > 255 not parsed correctly

efisher, 2019-06-06:
At 847ac29

When decoding a message such as:
```
message SimpleMessage {
  optional uint32 number = 1;
  extensions 100 to max;
}

message ExtMessage {
  extend SimpleMessage {
    optional ExtMessage ext_message_ext = 1100;
  }
  optional uint32 second_number = 1;
}
```

the `found` field is never set to true. However, If I switch `ext_message_ext = 200;` it is set correctly.

PetteriAimonen, 2019-06-07:
Hmm, for testing, can you uncomment this line from `pb.h` and see if works then:

     /* #define PB_FIELDINFO_WIDTH 4 */

If it does, it would point towards that the automatic field info width selection doesn't work correctly for extension fields for some reason.

PetteriAimonen, 2019-12-18:
I added a test case for this, but couldn't reproduce the issue, not on current master nor on 847ac29.

AnsgarSchmidt, 2019-12-29:
Hi,
Have the same problem here.
        [....]
        uint32                      monitor_ttg                                      =  208;
        float                       historic_depth_deepest_discharge                 =  300;
        float                       historic_depth_last_discharge                    =  301;
        [....]

And when I decode the protobuf with
curl http://192.168.4.178/bmv.protobuf/1 2>/dev/null   | protoc --decode odcore.VE_BMV_Data_Struct_TypeDef --proto_path=/host odcore_bmv.proto

monitor_ttg: 65535
44: 0xc25e6666
49: 18446744073709547619
50: 0x414a3d71
51: 0x41a251ec

Compiler call includes
-DPB_FIELD_16BIT \     // Worked with the older version pre 0.4.0
-DPB_FIELDINFO_WIDTH=4 \

Thank you
Ansgar

PetteriAimonen, 2019-12-29:
@AnsgarSchmidt Hmm, interesting. Are you using extensions or just normal fields? Which processor architecture are you using?

PetteriAimonen, 2019-12-29:
Also, if you have a debugger available, the following information could help in debugging:

- Value of `sizeof(int)`
- dump of the `MyMessage_field_info` array
- breakpoint at `pb_enc_fixed()` and print out contents of `*field`. Repeat until you get to one of the `field->tag` values that is being handled incorrectly.

AnsgarSchmidt, 2019-12-29:
> @AnsgarSchmidt Hmm, interesting. Are you using extensions or just normal fields? Which processor architecture are you using?

In this case normal fields.
Processor is an stm32F7 with gcc

Greetings
Ansgar

AnsgarSchmidt, 2019-12-29:
> Also, if you have a debugger available, the following information could help in debugging:
>
> * Value of `sizeof(int)`
> * dump of the `MyMessage_field_info` array
> * breakpoint at `pb_enc_fixed()` and [...]

Will give it a try later today and send you the data.

PetteriAimonen, 2019-12-29:
@AnsgarSchmidt Hmm, I think I found the bug you were hitting. The 8th bit of the tag number was being set to 0 because of a mistake in a bit mask in descriptor width 2 case.

The tests weren't catching this, as many of them test nanopb encoder -> nanopb decoder, and the same corruption was on both sides (#474). I now altered the `AllTypes` test case which compares against `protoc` to include larger tag numbers also, to cover this case.

Also looks like `PB_FIELDINFO_WIDTH` isn't actually overriding the width when the generator has already expanded the width (#473). Setting `(nanopb).descriptorsize=DS_4` in `.proto` options should work as a workaround in 0.4.0.

This does not directly match the situation @efisher was hitting, as his tag number 1100 already had 8th bit cleared.

AnsgarSchmidt, 2019-12-29:
So you don't need a the debug data?
Let me know when I can help with testing a fix.

Greetings
Ansgar

PetteriAimonen, 2019-12-30:
@AnsgarSchmidt  You can test now, fix is in revision 7c90bee403

PetteriAimonen, 2020-01-16:
@AnsgarSchmidt Have you had a chance to try if the fix helps for you?

PetteriAimonen, 2020-02-01:
I'm marking this as fixed now due to lack of feedback.

closed 2020-02-02 by PetteriAimonen