nanopb/nanopb #408: Linker issue - environmental?

agrath, 2019-06-22:
Pretty sure this is an environmental issue but I'm stuck and I don't know how to fix this!

I'm building using vscode, with arduino 1.8.9 for esp8266 using the 2.5.0 board library.

I've created a proto3 .proto file and compiled it with  nanopb-0.3.9.3 protoc to generate a packet.pb.c and packet.pb.h file

All the references to types and functions are working within the IDE, and the code compiles but when it comes to linking, I get this error:

`~/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: sketch\Main.ino.cpp.o:(.text._Z10sendPacket13_pb_Packet+0x0): undefined reference to `pb_Packet_fields'` (which is pb_field_t pb_Packet_fields[6]; in Packet.pb.h)

If I remove the extern const from the line where that is defined in the generated .h then all compiles/links fine.

Not sure why this is a problem, hopefully I've provided enough versions/detail!

PetteriAimonen, 2019-06-24:
Hi,

> `~/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: [...]

> If I remove the extern const from the line where that is defined in 
> the generated .h then all compiles/links fine.

Hmm, sounds interesting. I don't have time to try the compilation myself 
right now, but here are some ideas how you could debug it further:

1) Check `nm` symbol list on the files:
nm sketch\Main.ino.cpp.o     <-- Should have: U pb_Packet_fields
nm sketch\Packet.pb.o        <-- Should have: D pb_Packet_fields

2) Check if you can get to see the complete list of arguments that goes 
to 'ld' command and try it from the command line. Check that Packet.pb.o 
is there.

--
Petteri

agrath, 2019-06-26:
Thanks for that @PetteriAimonen 
This is a personal project so i'm only working on it nights and weekends but I'll check your suggestions ASAP - I really appreciate your help.
I pushed on with the `extern const` removed and have now observed that when I encode the packet to a stream with the following code then I get bytes_written as 1 (the varint size header) and the whole buffer is 0x0. Without using pb_encode_delimited (e.g. just pb_encode) then bytes_written is 0 and the whole buffer is 0x0.
I can't see anything wrong in my serialization/encode and the packet appears to be valid as well as my use of pointers so this makes me suspicious about the above issue with the fields struct.

Encode/send:
```
pb_Packet packet;
void sendPacket()
{
  Debug(F("sendPacket"));
  protoBuffer[0] = 0;
  pb_ostream_t stream = pb_ostream_from_buffer(protoBuffer, sizeof(protoBuffer));
  if (!pb_encode_delimited(&stream, pb_Packet_fields, &packet))
  {
    Debug(F("Encode of MQTT packet failed"));
    Debug(PB_GET_ERROR(&stream));
    return;
  }
  DebugF((const char *)F("Publishing %d bytes as MQTT to %s%s\n"), stream.bytes_written, MQTT_SERVER, MQTT_TOPIC);
  if (debugEnabled)
  {
    sharedCharacterBuffer[0] = 0;
    for (int i = 0; i < pb_Packet_size + pb_header_size; ++i)
    {
      sprintf(sharedCharacterBuffer, (const char *)F("%s%02X"), sharedCharacterBuffer, protoBuffer[i]);
    }
    DebugF((const char *)F("HEX encoded packet: %s\n"), sharedCharacterBuffer);
  }
  //MQTT publish etc, dropped for brevity
}
```

Packet build:
```
int32_t mac;
time_t utc;
void buildPacket()
{
  packet = pb_Packet_init_zero;
  packet.address = mac;
  packet.timestamp = utc;
  packet.battery = 3452;   //todo: actual mV
}
```

agrath, 2019-06-27:
[build-log.txt](https://github.com/nanopb/nanopb/files/3333293/build-log.txt)

Attached is my verbose build-log. I have edited it for brevity replacing my c:\users\profile folder with ~/profile/ and my project path with ~/project

The LD command on line 160 does not appear to reference Packet.pb.o, looking in the sketch subfolder of build, it's not there either (but the raw cpp/h files are)

I moved the generated .proto, .cpp and .h files out of a proto subfolder into the same level as main.ino and altered my include to match and the linker error with extern const is now gone. 

Unfortunately, I'm getting a different build error with an Arduino library that says it can't be found now.. and this is definitely not a nanopb issue.

I will leave this issue open until I can validate if this re-pathing has fixed the 0x0 buffer error

agrath, 2019-06-27:
Er, this is kinda weird. A bit of an update.
If I remove nanopb (all headers, code references etc) and my generated proto.cpp and .h files from my code folder, then my .ino builds correctly (all libraries resolved and linked correctly)
This issue didn't occur when the generated packet files were in a proto subfolder of the sketch, but i had the other issue with extern const & 0x0
If I bring nanopb and the generated proto definition back in, then when trying to compile, it crashes out on WifiManager (one of the arduino libs as being not found)
In the ~/Build/Sketch folder, I can only see Packet.pb.c.cpp.o if nanopb/proto is included, if it is excluded all the other .o files are compiled then main.ino.o is written
I am so confused - can anyone make any further suggestions?

agrath, 2019-06-27:
[log-withoutnanopb.txt](https://github.com/nanopb/nanopb/files/3334097/log-withoutnanopb.txt)
[log-withnanopb.txt](https://github.com/nanopb/nanopb/files/3334098/log-withnanopb.txt)

It appears that vscode/arduino plugin is finds the first .cpp/.ino file in the sketch folder and treats that as the one to compile as the root, so if the generated proto files are at the same level (which fixes my first issue) but alphabetically before the main ino, then compilation fails.

I renamed the .proto to be ZPacket.proto (generating ZPacket.pb.c) and now full compilation succeeds right through to generating the binary!

I can also now confirm that this whole exercise has also fixed the issue where the buffer was generated as 0x0, buffers are now being generated correctly. 

I won't close this issue myself in case you want to use this to debug the issue further but I don't know if it's strictly an issue with nanopb, though this is likely to be an issue for other people too.

Here is a log with a successful compilation: [log-success.txt](https://github.com/nanopb/nanopb/files/3334129/log-success.txt)

You can close this issue if you think this is beyond fixable.

closed 2019-06-30 by agrath