nanopb/nanopb #414: -L flag does not work as intended

ArjunNarayan2066, 2019-07-29:
When I use the `-L` or `--library-include-format=_some_folder/pb.h`, I expect in my `.pb.h` file to have `#include <_some_folder/pb.h`. However the script generated only `_some_folder/pb.h` for the include, not wrapped in `#include <>`. When I try to instead to `--library-include-format=\"#include <_some_folder/pb.h>\", I get various parsing errors with nanopb & bash. Is this the intended behaviour of this flag? Becuase that wouldn't make much sense. Let me know if theres any code/other info you all need.

PetteriAimonen, 2019-07-29:
The default value is `#include <%s>` so you should follow a similar format. Try using single apostrophes ' to wrap the value in bash.

That should work for the generated .pb.h files. The main nanopb core will however always need to be compiled with pb.h in include path.

ArjunNarayan2066, 2019-07-31:
Ive been testing quite a bit and still does not seem to work correctly when I give it a space. The space inside `#include <some/folder/to/nanopb/pb.h` it causing problems. Ive tried the following formats and they either show up as empty strings in the python script, or throw parsing errors.
```
--nanopb_out=-v,--library-include-format=`#include <some/folder/to/nanopb/%s>\n`:.
--nanopb_out=-v,--library-include-format="#include <some/folder/to/nanopb/%s>\n":.
--nanopb_out=-v,--library-include-format=\"#include <some/folder/to/nanopb/%s>\n\":.
--nanopb_out=-v,--library-include-format='#include <some/folder/to/nanopb/%s>\n':.
--nanopb_out=-v,--library-include-format=\`#include <some/folder/to/nanopb/%s>\n\':.
```
Is there something I am missing? I am running on an Ubuntu environment using zsh but the same command must work on MacOS.

PetteriAimonen, 2019-08-01:
Umm yeah, looks like it is either completely broken when used as protoc plugin, or otherwise it requires some crazy escaping that I couldn't figure out either.

As a workaround, you can do the generation in two steps:
<pre>
petteri@oddish:~/tmp4$ ~/nanopb-test/nanopb-0.3.9.3-linux-x86/generator-bin/protoc -ofoo.pb foo.proto 
petteri@oddish:~/tmp4$ ~/nanopb-test/nanopb-0.3.9.3-linux-x86/generator-bin/nanopb_generator -v --library-include-format="#include &lt;dir/%s>" foo.pb
petteri@oddish:~/tmp4$ grep include foo.pb.h
#include &lt;dir/pb.h>
</pre>

The options format is a bit less annoying when not passed through protoc.

But the option parsing through protoc is still a valid issue. I think #343 may be related.

PetteriAimonen, 2019-12-16:
Partially fixed by #450, but maybe not completely.

PetteriAimonen, 2019-12-20:
Passing options through `protoc` remains a bit difficult. The `--nanopb_opt=` argument syntax supported in protoc 3.6 and newer makes this a bit easier.

In addition, 0.4.0 now allows just passing .proto files to `nanopb_generator.py` directly, which makes it much easier to pass options to it.

closed 2019-12-20 by PetteriAimonen