nanopb/nanopb #415: Encoded size macro fails to compile on c++

muratsevim, 2019-07-31:
For some proto messages including oneof fields, the encoded size macro is generated but it fails to compile when used in c++ code. I get the error below when trying to use it:

foo.pb.h:52:67: error: types may not be defined in ‘sizeof’ expressions
 #define Foo_size                                 (5 + sizeof(union{char f1[Bar_size]; char f2[Bar_size];}))

Attached example proto files I used to reproduce the issue.

[foo_proto.zip](https://github.com/nanopb/nanopb/files/3451088/foo_proto.zip)

PetteriAimonen, 2019-07-31:
There are two things happening here.

1. For some reason, nanopb generator doesn't find bar.proto when compiling foo.proto. That may be because they are in different subdirectories (#116). You can try passing separate include path to nanopb generator like `--nanopb_out=-Isubdir:outputdir`. If the generator can find the file, it will produce a simple `#define Foo_size  146`.

2. C++ doesn't support the trick used in #338 for computing the size at compile time. I guess one could `#ifdef` a C++-specific alternative to that, but because it is just a fallback anyway, I think fixing include paths is a better solution.

Here's what happens for me when all files are in same directory:
<pre>
petteri@oddish:~/tmp4$ ~/nanopb-test/nanopb-0.3.9.3-linux-x86/generator-bin/protoc --nanopb_out=. bar.proto
petteri@oddish:~/tmp4$ ~/nanopb-test/nanopb-0.3.9.3-linux-x86/generator-bin/protoc --nanopb_out=. foo.proto
petteri@oddish:~/tmp4$ grep Foo_size foo.pb.h
#define Foo_size                                 146
</pre>

muratsevim, 2019-07-31:
Thanks for the help. Looks like this happens only when I use --proto_path flag. Adding the include path flag as well works!

closed 2019-12-16 by PetteriAimonen