nanopb/nanopb #417: Failed to import the protocol definitions for generator

ElectricRCAircraftGuy, 2019-08-01:
I git cloned the repo but can't seem to be able to build anything.

```
$ protoc --plugin=protoc-gen-nanopb=/home/gabriels/GS/dev/Protocol_Buffers/Nanopb/source/nanopb/generator/protoc-gen-nanopb --nanopb_out=. simple.proto 

         ********************************************************************
         *** Failed to import the protocol definitions for generator.     ***
         *** You have to run 'make' in the nanopb/generator/proto folder. ***
         ********************************************************************
    
Traceback (most recent call last):
  File "/home/gabriels/GS/dev/Protocol_Buffers/Nanopb/source/nanopb/generator/nanopb_generator.py", line 39, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/home/gabriels/GS/dev/Protocol_Buffers/Nanopb/source/nanopb/generator/proto/nanopb_pb2.py", line 11, in <module>
    from google.protobuf import symbol_database as _symbol_database
ImportError: cannot import name symbol_database
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

Here's my output from a couple things asked for here: https://github.com/nanopb/nanopb/issues/240#issuecomment-277238267

```
$ which protoc
/usr/local/bin/protoc
```

```
$ protoc --version
libprotoc 3.9.0
```

```
$ python -c 'import google.protobuf; print(google.protobuf.__file__)'
/usr/lib/python2.7/dist-packages/google/protobuf/__init__.pyc
```

PetteriAimonen, 2019-08-02:
Hi,

> ```
> $ which protoc
> /usr/local/bin/protoc
> ```
> ```
> $ python -c 'import google.protobuf; print(google.protobuf.__file__)'
> /usr/lib/python2.7/dist-packages/google/protobuf/__init__.pyc
> ```

It seems like you have a newer protoc installed under 
/usr/local/bin/protoc, but the Python protobuf library is the one that 
comes with your distribution.

You can try to use /usr/bin/protoc instead:
PATH=/usr/bin make -BC generator/proto

The protobuf python library is a bit annoying in that the protoc version 
needs to match with the python library version. Hopefully the protoc 
under /usr/bin will match.

--
Petteri

ElectricRCAircraftGuy, 2019-08-05:
Here's what I got when I tried to build, per your recommendation:

```
~/GS/dev/Protocol_Buffers/Nanopb/source/nanopb $ PATH=/usr/bin make -BC generator/proto
make: Entering directory `/home/gabriels/GS/dev/Protocol_Buffers/Nanopb/source/nanopb/generator/proto'
protoc --python_out=. nanopb.proto
google/protobuf/descriptor.proto:428:3: Expected "required", "optional", or "repeated".
google/protobuf/descriptor.proto:428:12: Expected field name.
google/protobuf/descriptor.proto:486:3: Expected "required", "optional", or "repeated".
google/protobuf/descriptor.proto:486:12: Expected field name.
google/protobuf/descriptor.proto:487:3: Expected "required", "optional", or "repeated".
google/protobuf/descriptor.proto:487:12: Expected field name.
google/protobuf/descriptor.proto:586:3: Expected "required", "optional", or "repeated".
google/protobuf/descriptor.proto:586:12: Expected field name.
google/protobuf/descriptor.proto:609:3: Expected "required", "optional", or "repeated".
google/protobuf/descriptor.proto:609:12: Expected field name.
nanopb.proto: Import "google/protobuf/descriptor.proto" was not found or had errors.
nanopb.proto:133:8: "google.protobuf.FileOptions" is not defined.
nanopb.proto:137:8: "google.protobuf.MessageOptions" is not defined.
nanopb.proto:141:8: "google.protobuf.EnumOptions" is not defined.
nanopb.proto:145:8: "google.protobuf.FieldOptions" is not defined.
make: *** [nanopb_pb2.py] Error 1
make: Leaving directory `/home/gabriels/GS/dev/Protocol_Buffers/Nanopb/source/nanopb/generator/proto'
```

ElectricRCAircraftGuy, 2019-08-05:
Each of these errors (ex: `google/protobuf/descriptor.proto:609:3: Expected "required", "optional", or "repeated".`) seems to be because it doesn't like the `reserved` keyword, and is saying it must instead be `required`, `optional`, or `repeated`. 

I'm a little confused why it won't accept `reserved` and requires the deprecated proto2 `required` or `optional`. Is there a proto3-compatible version of nanopb?

ElectricRCAircraftGuy, 2019-08-05:
Cross-reference: https://stackoverflow.com/questions/57367265/how-to-compile-nanopb-proto-file-into-h-and-c-files-using-nanopb-and-protobuf

PetteriAimonen, 2019-08-06:
Ok, it appears like your /usr/bin/protoc is so old that it doesn't support proto3 format.
Because your /usr/local/bin/protoc is up to date, it might be easiest if you also install protobuf-python of the 3.9.0 version. Note that the protoc version and the protobuf-python version have to match.

Nanopb does support proto3 syntax, it's just that you have two conflicting versions of Google's protoc compiler on your system, and one too old python protobuf library.

You can also use one of the binary packages that already come with these dependencies.

ElectricRCAircraftGuy, 2019-08-06:
Thanks. I got it to work. I had built the protobuf library from source, but I had only followed the C++ build instructions: https://github.com/protocolbuffers/protobuf/tree/master/src

By following the Python build instructions too (https://github.com/protocolbuffers/protobuf/tree/master/python), it worked. 

See full details in my answer here: https://stackoverflow.com/questions/57367265/how-to-compile-nanopb-proto-file-into-h-and-c-files-using-nanopb-and-protobuf/57367543#57367543.

ElectricRCAircraftGuy, 2019-08-06:
Pull request: https://github.com/nanopb/nanopb/pull/418

I'm reopening this issue until you address (and hopefully accept) my pull request to add clarity to the readme.

ElectricRCAircraftGuy, 2019-08-08:
Thanks for merging my readme change pull request.

closed 2019-08-08 by ElectricRCAircraftGuy