nanopb/nanopb #419: nanopb version 0.3.9.3 does not work on Windows

andrew-mclachlan, 2019-08-16:
nanopb-windows-x86\generator-bin>protoc.exe --nanopb_out=source --proto_path=myfile.proto

         *************************************************************
         *** Could not import the Google protobuf Python libraries ***
         *** Try installing package 'python-protobuf' or similar.  ***
         *************************************************************

Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__protoc-gen-nanopb__.py", line 23, in <module>
ImportError: No module named protobuf.text_format
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

PetteriAimonen, 2019-08-16:
Hmm, works for me.

Can you try to run the following commands in command prompt in the generator-bin directory:

<pre>

C:\...\generator-bin> where protoc-gen-nanopb
C:\Users\Administrator\Desktop\nanopb-0.3.9.3-windows-x86\generator-bin\protoc-gen-nanopb.exe

C:\...\generator-bin> py.exe
Python 2.7.5 (default, May 15 2013, 22:43:36) [MSC v.1500 32 bit (Intel)] on win32

>>> import google
>>> print(google.__file__)
c:\users\administrator\desktop\nanopb-0.3.9.3-windows-x86\generator-bin\protobuf-3.6.1-py2.7.egg\google\__init__.pyc

>>> import google.protobuf.text_format
>>> print(google.protobuf.text_format.__file__)
c:\users\administrator\desktop\nanopb-0.3.9.3-windows-x86\generator-bin\protobuf-3.6.1-py2.7.egg\google\protobuf\text_format.pyc
</pre>

Please paste the results. Also, what version of Windows are you using?

andrew-mclachlan, 2019-08-16:
```
nanopb-windows-x86\generator>where protoc-gen-nanopb
nanopb-windows-x86\generator\protoc-gen-nanopb
nanopb-windows-x86\generator\protoc-gen-nanopb.bat

nanopb-windows-x86\generator>py
Python 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 23 2018, 23:31:17) [MSC v.1916 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import google
>>> print(google.__file__)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'google' has no attribute '__file__'
```
I'm using Windows 10. I ran `pip install protobuf` too.

andrew-mclachlan, 2019-08-16:
```
nanopb-windows-x86\generator-bin>.\nanopb_generator.exe

         *************************************************************
         *** Could not import the Google protobuf Python libraries ***
         *** Try installing package 'python-protobuf' or similar.  ***
         *************************************************************

Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__nanopb_generator__.py", line 23, in <module>
ImportError: No module named protobuf.text_format
```

PetteriAimonen, 2019-08-16:
Hm, it looks like you ran the commands in generator (not generator-bin) folder.
But I guess I got most of the information I wanted anyway.

I retested on Windows 10, but it also works for me.

Could you check if you have some `python27.dll` in your Windows system folders?
I'm not sure how that would happen, normal python installation shouldn't put that there, but after all you have `py.exe` there also.

You can also try dependency walker (from http://www.dependencywalker.com/ use x86 version because this is 32-bit binary) to check where `nanopb_generator.exe` is trying to load libraries from. It will give some errors about implicit / forwarded depedencies, but it will show where the direct dependency `python27.dll` is coming from if you right click and select "show full paths".

andrew-mclachlan, 2019-08-16:
I made sure that I removed all my Python installation paths from my environment path(s).
I ran dependency walker and it revealed: `c:\temp\nanopb\nanopb-windows-x86\generator-bin\PYTHON27.DLL`. I'm using Windows 10 64-bit.

Is there any tracing that I can get out? Thank you for your support btw!! Most appreciated!!

PetteriAimonen, 2019-08-17:
Hmm, I'm quite out of ideas. Normally even existing python installations shouldn't affect the binary packages.

I'm quite out of ideas. Trying the import from the `py.exe` in generator-bin folder might reveal something more, but it might not.

One final thing you could double-check is run `dir` in generator-bin folder and check that all files are present. In particular the two `.egg` directories and `library.zip`:

<pre>
17.08.2019  08:34    <DIR>          .
17.08.2019  08:34    <DIR>          ..
16.08.2019  16:56            68 608 bz2.pyd
16.08.2019  16:56           740 822 library.zip
16.08.2019  16:56             1 850 Microsoft.VC90.CRT.manifest
16.08.2019  16:56           421 200 MSVCP100.DLL
16.08.2019  16:56           770 384 MSVCR100.DLL
16.08.2019  16:56           653 136 MSVCR90.DLL
16.08.2019  16:57                13 myfile.proto
16.08.2019  16:56             7 680 nanopb_generator.exe
16.08.2019  16:56    <DIR>          protobuf-3.6.1-py2.7.egg
16.08.2019  16:56             7 680 protoc-gen-nanopb.exe
16.08.2019  16:56         2 437 120 protoc.exe
16.08.2019  16:56             7 680 py.exe
16.08.2019  16:56           127 488 pyexpat.pyd
16.08.2019  16:56         2 448 384 python27.dll
16.08.2019  16:56            10 240 select.pyd
16.08.2019  16:56           192 696 setuptools-0.6c11-py2.7.egg
16.08.2019  16:56    <DIR>          six-1.10.0-py2.7.egg
16.08.2019  16:56           686 080 unicodedata.pyd
16.08.2019  16:56           358 400 _hashlib.pyd
16.08.2019  16:56            44 544 _socket.pyd
16.08.2019  16:56           899 072 _ssl.pyd
              19 File(s)      9 883 077 bytes
               4 Dir(s)  17 647 353 856 bytes free
</pre>

andrew-mclachlan, 2019-08-17:
Working now!
I think my issue was down to the fact that I had downloaded the windows zip on macOS, extracted it on macOS and then using it from my Windows VM.
When I re-downloaded the zip file (this time on Windows), it worked well!

Thanks for your support. I think we can close this Issue as Not a bug! Thanks!!

andrew-mclachlan, 2019-08-17:
The gotcha! is that I had committed the contents of the zip file for Windows to git and it had ignored directories in the .egg directories, such as `google`!
It worked on other platforms due to the difference in the generator-bin directory content.

PetteriAimonen, 2019-08-17:
Ah, good that you found the reason.

I've now added a rule so that generator-bin will not be excluded. As for the other files (such as under `generator/proto`), it is a bit more complex, as there are files that should be ignored under normal development flow even though they are included in binary package. Also there are some `.pb` files under tests that get matched by gitignore currently but are included in the repo.

closed 2019-12-20 by PetteriAimonen