nanopb/nanopb #420: problem with callbacks

rabad0001, 2019-08-19:
Hi all, I have the following protobufs (code has been edited for public form):

// update_data.proto
```
syntax = "proto3";

message UpdateAppData {
  uint32 app_data_size = 1;
  bytes app_data = 2;
}
```

// request.proto
```
syntax = "proto3";

import "update_data.proto";

enum RequestId {
  // other request Ids

  UPDATE_APP_DATA_REQ = 9995;

  // more request Ids
}

message Request {
  RequestId request_id = 2;

  oneof parameters {
    // other stuff

    // use with TSB_FWUPDATE_APP_DATA_REQ
    UpdateAppData update_app_data = 3;

    // more stuff
  }
}

```
// payload.proto
```
syntax = "proto3";

import "request.proto";

message PayloadHeader {
  uint32 transaction_id = 1;
}

message Message {
  oneof request_response {
    Request request = 1;

    // more stuff
  }
}

message Payload {
  PayloadHeader payload_header = 1;
  repeated Message messages = 2;
}

```
The callback to decode the bytes type in UpdateAppData is:
```
 `static bool TsbMessaging_DecodeUpdateAppDataMsg(pb_istream_t *stream,
                                                     const pb_field_t *field,
                                                     void **arg) {
  uint8_t *data_buffer = Update_GetBuffer();
  uint32_t count = stream->bytes_left;
  if (stream->bytes_left > MAX_UPDATE_BUFFER_SZ) return false;

  if (!pb_read(stream, data_buffer, stream->bytes_left)) return false;
  return true;
}

```
So when my application receives a Payload, the following callback gets called:

```
Message msg;
static bool Messaging_DecodeRepeatedSubmsg(pb_istream_t *stream,
                                              const pb_field_t *field,
                                              void **arg) {

  uint32_t update_size =
      msg.request_response.request.update_app_data.app_data_size;
  msg.request_response.request.update_app_data.app_data.arg =
      &update_size;
  msg.request_response.request.update_app_data.app_data.funcs.decode = &Messaging_DecodeTxxUpdateAppDataMsg;

  /* Decode message into sub message array */
  return pb_decode(stream, Message_fields, &msg);
}
```

However, when pb_decode() gets called by this callback, the following snippet gets executed in pb_decode.c(decode_static_field()) when processing Request:
```
        case PB_HTYPE_ONEOF:
            *(pb_size_t*)iter->pSize = iter->pos->tag;
            if (PB_LTYPE(type) == PB_LTYPE_SUBMESSAGE)
            {
                /* We memset to zero so that any callbacks are set to NULL.
                 * Then set any default values. */
                memset(iter->pData, 0, iter->pos->data_size);
                pb_message_set_to_defaults((const pb_field_t*)iter->pos->ptr, iter->pData);
            }
            return func(stream, iter->pos, iter->pData);

```

When this snippet executes, it erases the callback I had set for the bytes type in UpdateAppData.  So the bytes never get decoded.

So I have the following questions:
1) I'm setting up the callback for UpdateAppData in the callback for Payload.  Is this the proper place to be doing that?
2) Assuming I'm setting up the callback for UpdateAppData properly, what can I do to prevent the call to decode_static_field() from clobbering it?
3) When the bytes callback gets erased, pb_skip_field() is called instead of my desired bytes callback.  Is there a way to recover this skipped field so that I could just simply do my own decoding when pb_decode() returns??

Any insights you can offer would be greatly appreciated.  Thanks in advance!

- Robert

PetteriAimonen, 2019-08-20:
Sounds like a duplicate of #175.

Because oneofs are made into C unions, they share the same memory so there is no place to store callbacks separately.
The easiest workaround is to specify `(nanopb).no_unions = true` so that each field inside oneof gets its own storage.

rabad0001, 2019-08-20:
Petteri, thank you for the response.

How would I do this using an options file?
// request.options
Request.paramters no_unions:true

???

PetteriAimonen, 2019-08-20:
Yeah, though you'd want to do that for `Message.request_response` also. You can also use `* no_unions:true` to apply it for all oneofs by default.

PetteriAimonen, 2019-08-27:
It's nanopb only. There is the protobuf feature of specifying options in .proto also, which nanopb supports. But some other protobuf libraries were confused by the extra options, which is why the separate .options file was added as an alternative.

deviousasti, 2019-09-10:
I've got several messages in the oneof, so if I go with the no unions, my stack will blow up.
I hope the callback erasure will be mitigated with `setup_oneof` or the like.

PetteriAimonen, 2019-12-16:
Closing in favor of #175.

closed 2019-12-16 by PetteriAimonen