nanopb/nanopb #421: Socket streaming with zero-length string fields

michael-vacek, 2019-08-20:
Hi all,

I observed some strange difference in behavior when decoding message from fixed-length buffer (pb_istream_from_buffer) and from streaming socket (pb_istream_from_socket in examples/network_server/common.c).

When the message contains empty string, the latter case causes to wait for additional data even all needed data were already received. If I use the same data with pb_istream_from_buffer, the return status is true and the message struct is assigned correctly.

This happens only if the message has empty string (0x0).

Is this behavior expected?

Best regards,
Michael

PetteriAimonen, 2019-08-21:
Which are you using, `pb_decode_delimited()` or `pb_decode()`?
Plain `pb_decode()` doesn't know when message has ended because there is no length info in plain protobuf messages.

Not sure why empty string would affect it.

michael-vacek, 2019-08-21:
Hello Peter,

I use `pb_decode_delimited()`. On the server-side, I use almost identical code you have in `examples/network_server/common.c` just with delimited decoding. On the client-side, I use python code generated from protoc and before the actual message, I send a varint with the message length. As already said, this works ok, until some of the `optional string` field is present and set to `""`.

I attach the core of my proto message just for the case you would have an idea what is wrong.

```
message msg_m
  {
    optional action_t action = 1 [default = READ];
    optional comm_if_t src_if = 2;
    oneof values
    {
        msg_type_1 msg_type_1 = 3;
        msg_type_2 msg_type_1 = 4;
        ...
    }
    optional string name = 10;
  }
````

and my options are:
```
* max_size:40
* max_count:5
* fbytes fixed_length:true max_size:4
* no_unions:true;
* anonymous_oneof:true;
```

Alternatively, I will collect the messages from a socket into a buffer and use `pb_istream_from_buffer` instead which do not exhibit this behavior.

Best regards,
Michael

PetteriAimonen, 2019-08-22:
`I use almost identical code you have in examples/network_server/common.c just with delimited decoding.`

Hmm, what do you mean by that? The network_server example already uses `pb_decode_delimited()` in the client & server code, and there should be no reason to modify the socket code for that.

PetteriAimonen, 2019-08-22:
Hmm, I just now noticed that the code may call `recv()` with len set to 0 when there is an empty string. You could try adding `if (count > 0)` in `read_callback()`.

PetteriAimonen, 2019-08-22:
Seems actually like a valid bug in the example, because with `count = 0` it will falsely detect EOF condition when `recv()` returns 0.

michael-vacek, 2019-08-22:
Thanks for your help.

PetteriAimonen, 2019-10-13:
Fix released in 0.3.9.4.

closed 2019-10-13 by PetteriAimonen