nanopb/nanopb #425: Which one of the `oneof` when `no_union=true`

oliviermartin, 2019-08-26:
To workaround the decode callback issue when in a `oneof` (see for instance the last issue https://github.com/nanopb/nanopb/issues/420), it is recommended to use `no_unions = true`.

When using this nanopb option, the generated structure does not expose the `which_` attribute allowing to know which `oneof` was called.

Here to duplicate the limitation:

```
syntax = "proto3";

import "nanopb.proto";

message RecordServerMessage {
  uint32 sequence = 1;
}

message RecordServerMessages {
  repeated RecordServerMessage records = 1;
}

message ServerMessage {
  option (nanopb_msgopt).no_unions = true;
  oneof message {
    RecordServerMessage record_server = 1;
    RecordServerMessages record_servers = 2;
  }
}
```

Generate the code with: `protoc --plugin=nanopb/generator/protoc-gen-nanopb --nanopb_out=. --proto_path=/tmp/ --proto_path=nanopb/generator/proto /tmp/no_union.proto`

And here is the generated structure:

```
typedef struct _ServerMessage {
    RecordServerMessage record_server;
    RecordServerMessages record_servers;
/* @@protoc_insertion_point(struct:ServerMessage) */
} ServerMessage;

```

A workaround could be to use decode callbacks for all the fields of the `oneof` structure.

But keeping the `which_` attribute would make the code much simpler and readable when some attributes do not need to the complexity of the decoding callback.

To clarify it is this generated `which_message` I mean when `no_union=false`:

```
typedef struct _ServerMessage {
    pb_size_t which_message;
    union {
        RecordServerMessage record_server;
        RecordServerMessages record_servers;
    } message;
/* @@protoc_insertion_point(struct:ServerMessage) */
} ServerMessage;
```

PetteriAimonen, 2019-08-26:
Hmm indeed. For proto2 it generates the `has_` fields, but it would make sense to generate `which_` instead for proto3.

PetteriAimonen, 2019-10-02:
As a workaround, one can specify `no_unions:true, proto3:false` to get separate `has_` fields for each field, but that is hardly ideal.

I think I'm not going to make a fix for this in the 0.3.9 series, though it would be nice to finally figure out something useful for the callback issue in 0.4.0.

PetteriAimonen, 2019-12-20:
Now in 0.4.0, there are more options for dealing with callbacks inside oneofs (see tests/oneof_callbacks).

In addition, proto3 submessages now also have `has_` field, providing ability to identify them even when `no_unions` is set.

closed 2019-12-20 by PetteriAimonen