nanopb/nanopb #431: Can't install in PlatformIO (Mac)

xoxota99, 2019-09-16:
I'm getting the following error when attempting to install: 

PIO Core Call Error: "Library Storage: /Users/pdesrosiers/.platformio/lib\nLibraryManager: Installing id=431 @ ^0.3.9.2\n\n\nError: Could not find a version that satisfies the requirement '^0.3.9.2' for your system 'darwin_x86_64'"

PetteriAimonen, 2019-09-16:
@ivankravets Do you have any idea about this?

selangh, 2019-10-28:
Same here on my Mac. I downloaded the nanopb repo (I use version 0.3.9 at the moment) into my projects "lib" dir --> `$PRJPATH/lib/nanopb-nanopb-0.3.9`. 

Then I created `pre-build.py`in the projects root:
```python
#!/usr/local/bin/python3
Import("env")

print("----------PRE-BUILD--------------")
env.Execute("which python")
env.Execute("pip install grpcio-tools")
env.Execute("cd lib/nanopb-nanopb-0.3.9/generator/proto/ && make")
env.Execute("rm -f src/*.pb.c")
env.Execute("rm -f include/*.pb.h")
env.Execute("protoc --plugin=protoc-gen-nanopb=lib/nanopb-nanopb-0.3.9/generator/protoc-gen-nanopb \
                    --nanopb_out=. src/*.proto")
env.Execute("mv src/*.pb.h include/")
```

To call this script a modified my `platformio.ini`
```bash
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_port = /dev/tty.wchusbserial14510
extra_scripts = pre:pre-build.py
```

Maybe not the smartest method but it works.

PetteriAimonen, 2019-12-20:
I'm not sure what I can do about this, it may be more fruitful to open it as an issue on platformio repo.

closed 2019-12-20 by PetteriAimonen