nanopb/nanopb #432: How to use cmake option 'NANOPB_OPTIONS'

oliviermartin, 2019-09-16:
I am trying to force the option `option (nanopb_msgopt).no_unions = true;` to all my `.proto` files.
I noticed the cmake variable `NANOPB_OPTIONS`.

I tried: `set(NANOPB_OPTIONS -s \(nanopb_msgopt\).no_unions:true)` but got the error message:
```
google.protobuf.text_format.ParseError: 1:1 : Expected identifier or number, got ;.
```

I also tried: `set(NANOPB_OPTIONS -s no_unions:true)` but also got the error message:
```
google.protobuf.text_format.ParseError: 1:1 : Expected identifier or number, got ;.
```

I also tried through the command line: `protoc --plugin=nanopb/generator/protoc-gen-nanopb --nanopb_out=-s\ no_unions:true,. --proto_path=../examples/test_simple/ --proto_path=nanopb/generator/proto  ../examples/test_simple/simple.proto`
and I also have error:
```
Traceback (most recent call last):
  File "nanopb/generator/nanopb_generator.py", line 1871, in <module>
    main_plugin()
  File "nanopb/generator/nanopb_generator.py", line 1850, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "nanopb/generator/nanopb_generator.py", line 1658, in parse_file
    text_format.Merge(s, toplevel_options)
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 536, in Merge
    descriptor_pool=descriptor_pool)
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 590, in MergeLines
    return parser.MergeLines(lines, message)
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 623, in MergeLines
    self._ParseOrMerge(lines, message)
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 638, in _ParseOrMerge
    self._MergeField(tokenizer, message)
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 749, in _MergeField
    tokenizer.Consume(':')
  File "/home/olivier/.local/lib/python2.7/site-packages/google/protobuf/text_format.py", line 1101, in Consume
    raise self.ParseError('Expected "%s".' % token)
google.protobuf.text_format.ParseError: 1:10 : Expected ":".
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2019-09-16:
Sounds like the issue is passing `:` through protoc, which itself uses `:` as separator between plugin options and path.

I don't currently have time to dig through protoc source code for an answer to that.

You could try running in two steps, that makes passing nanopb options much easier:

    protoc -ofoo.pb foo.proto
    python nanopb_generator.py -s no_unions:true foo.pb

oliviermartin, 2019-09-16:
For later reference, I am using a old `protoc` (v3.0.0 from Ubuntu 18.04 LTS) - and I have not tried yet with a newer version.

And I have just tried passing an additional parameter to `--nanopb_out` and I have a new error: 

```
protoc --plugin=nanopb/generator/protoc-gen-nanopb '--nanopb_out=-v,-s no_unions:true:.' --proto_path=../examples/test_simple/ --proto_path=nanopb/generator/proto  ../examples/test_simple/simple.proto 
... PARAMS: -v,-s no_unions
Usage: Usage: protoc --nanopb_out=[options][,more_options]:outdir file.proto

nanopb_generator.py: error: no such option: -,
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 2.
```

I am happy to raise this issue to protoc after trying the latest version. But I have not found where it is documented protobuf extension support argument passing through `.._out=list_args:OUT_DIR`

PetteriAimonen, 2019-09-16:
I'm not sure if it is even documented on protoc side.

sdorre, 2019-11-20:
Hello @oliviermartin 
You don't need any `,` to add options to `--nanopb_out` Juste let an empty space between and it should work. 

`protoc --plugin=nanopb/generator/protoc-gen-nanopb '--nanopb_out=-v -s no_unions:true:.' --proto_path=../examples/test_simple/ --proto_path=nanopb/generator/proto  ../examples/test_simple/simple.proto`

PetteriAimonen, 2019-12-20:
I think this should work now in 0.4.0. Please reopen if there are still issues.

closed 2019-12-20 by PetteriAimonen

bonastreyair, 2020-09-15:
I'm trying to use `'--nanopb_out=-v -s submsg_callback:true:.'` and the same error appears 

```
google.protobuf.text_format.ParseError: 1:16 : Expected ":".
```

I'm using protobuf version 3.13.

PetteriAimonen, 2020-09-15:
@bonastreyair Hmm yeah, the way it is fixed in CMake rules is using the (new, undocumented?) protoc option `--nanopb_opt`.

You can use it like this: `~/nanopb/generator/protoc '--nanopb_opt=-v -s submsg_callback:true' --nanopb_out=. test.proto`

bonastreyair, 2020-09-15:
Thank you so much, it works!

I think that adding this information in the documentation would help a lot to other users. 😃