nanopb/nanopb #433: A cyclic dependency exists

alrevuelta, 2019-09-26:
I am trying to generate the `pb-c.c` and `pb-c.h` files but I get the following error `AssertionError: A cyclic dependency exists amongst ...` and then a set of different messages that I have defined in the `.proto` file.

- I am running `nanopb-0.3.9.3-macosx-x86` binaries.
- The command that I'm running is `protoc --plugin=protoc-gen-nanopb=generator/protoc-gen-nanopb --nanopb_out=. myFile.proto`

There was a related issue open some time ago [https://github.com/nanopb/nanopb/issues/130](https://github.com/nanopb/nanopb/issues/130) but was closed and solved `Fix released in nanopb-0.3.1`. As I stated I'm using the binaries I downloaded, which are `0.3.9.3`

I am quite sure that the `proto` file I'm using is correct, since it comes from a trusted source. In fact, I could generate the `pb-c` files with `protoc` tool without any issue.

Any idea?
Thanks!

PetteriAimonen, 2019-09-27:
Are you trying to generate the fields as static submessages, or as callback or pointer fields?

Is there really a cyclic dependency in the .proto file?

alrevuelta, 2019-09-28:
The only custom made modification that I'm using is setting a `max_size` in the `.options` file for the strings. Without that I get `Exception: Callback fields inside of oneof are not supported (field dim_value)`. Should I specify static/callback/pointer type in my options file? Is this related to my issue?

Regarding the cyclic dependency I would say there is. Is there anything wrong with using `TypeProto` inside `TypeProto`?

```
message TypeProto {
  message Tensor {
    optional int32 elem_type = 1;
    optional TensorShapeProto shape = 2;
  }
  message Sequence {
    optional TypeProto elem_type = 1;
  };
```
I'm using this `.proto` [file](https://github.com/onnx/onnx/blob/master/onnx/onnx.proto) and the error I get is:
```
AssertionError: A cyclic dependency exists amongst {'onnx_ValueInfoProto': set(['onnx_TypeProto']), 'onnx_TypeProto_Map': set(['onnx_TypeProto']), 'onnx_TypeProto': set(['onnx_TypeProto_Sequence', 'onnx_TypeProto_Map']), 'onnx_TypeProto_Sequence': set(['onnx_TypeProto'])}
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2019-09-28:
Yeah, you'll need to specify type=FT_POINTER or type=FT_CALLBACK for the `elem_type` field.

Otherwise it would get generated as:

    struct TypeProto  {
        ...
        TypeProto elem_type;
    }

which wouldn't be legal in C.

alrevuelta, 2019-09-29:
Solved, thanks!

closed 2019-09-29 by alrevuelta