nanopb/nanopb #435: Coverity issue: Out-of-bounds read at file pb_encode.c

jsantana-broadsoft, 2019-10-04:
Hello,

We run Coverity analysis tool in our code base and a Out-of-bounds read issue was reported in file pb_encode.c file. 

I am attaching a screenshot of the issue details from the Coverity tool. I hope it provides enough details. 

[image]

Thanks!

PetteriAimonen, 2019-10-04:
That would only happen with invalid field descriptors (PB_HTYPE_ONEOF without pSize).

But the logic is needlessly complicated there, it has been cleaned up in current master branch.

jsantana-broadsoft, 2019-11-12:
Thank for the reply. I see the implementation of that method in master change substantially. 

We are not using it directly, it seems several Google frameworks are using it, Firebase specifically. For some reason, Firebase is using a 0.3.9011 version of nanopb.

I think we have to wait until they move up to a newer nanopb to get that issue resolved. 

Thanks!

closed 2019-11-12 by jsantana-broadsoft