nanopb/nanopb #437: Add check whether string field is a valid UTF-8 string

prusnak, 2019-10-10:
According to Protobuf specification, a string field should be a valid UTF-8 string. Currently, it is possible to receive a non-UTF-8 binary data when a string is expected.

Do we want to have such a check in the library? (It would be a compile-time option to enable this check, it would be disabled by default).

PetteriAimonen, 2019-12-16:
I guess it would be sensible to have the check as an option.

prusnak, 2019-12-16:
Would you be OK with patch like this?

``` diff
diff --git a/pb_decode.c b/pb_decode.c
index 62a7d97..6aa59af 100644
--- a/pb_decode.c
+++ b/pb_decode.c
@@ -1507,7 +1507,15 @@ static bool checkreturn pb_dec_string(pb_istream_t *stream, const pb_field_iter_
     }
     
     dest[size] = 0;
+
+#ifdef PB_TEST_STRING_UTF8
+    if (!pb_read(stream, dest, (size_t)size))
+        return false;
+
+    return test_valid_utf8(dest, size);
+#else
     return pb_read(stream, dest, (size_t)size);
+#endif
 }
 
 static bool checkreturn pb_dec_submessage(pb_istream_t *stream, const pb_field_iter_t *field)
```

Obviously it's missing `test_valid_utf8` function, but I just want to discuss whether it's OK to read a value from stream and validate it afterwards.

PetteriAimonen, 2019-12-16:
Yeah, that's ok. I'd bikeshed it this way:

    #ifdef PB_VALIDATE_UTF8
    if (!pb_read(stream, dest, (size_t)size))
        return false;

    if (!pb_validate_utf8(dest))
        PB_RETURN_ERROR(stream, "invalid utf8");

    return true
    #else
        ...
    #endif

but the details aren't that critical :)

There are probably proper references somewhere on how to parse & validate utf-8, but I'll just throw this link here of some old parsing code I've written: https://github.com/mcufont/mcufont/blob/master/decoder/mf_encoding.c

prusnak, 2019-12-16:
Created a draft PR in https://github.com/nanopb/nanopb/pull/465

closed 2019-12-20 by PetteriAimonen