nanopb/nanopb #440: nanopb 0.3.9.4 produces warnings using gcc's -Wbad-function-cast

cstender, 2019-10-16:
The new code introduced in nanopb 0.3.9.4 for proper handling of boolean fields procudes warnings when GCC's -Wbad-function-cast check is enabled.

```
-bash-4.2$ gcc -O2 -Wall -Wextra -Wconversion -Wformat-security -Wformat=2 -Wshadow -Wcast-qual -Wcast-align -Wredundant-decls -Wstrict-prototypes -Wbad-function-cast -g  -I include -I ../include -std=c99 -c -o obj/x86_64/pb_encode.o src/pb_encode.c
src/pb_encode.c: In function 'pb_enc_bool':
src/pb_encode.c:762:22: warning: cast from function call of type '_Bool' to non-matching type 'unsigned int' [-Wbad-function-cast]
     uint32_t value = (uint32_t)safe_read_bool(src);
                      ^
```

Code:
```
static bool checkreturn pb_enc_bool(pb_ostream_t *stream, const pb_field_t *field, const void *src)
{
    uint32_t value = (uint32_t)safe_read_bool(src);
    PB_UNUSED(field);
    return pb_encode_varint(stream, value);
}
```

The code looks fine and we think that this is a GCC issue (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=92117). Nevertheless we need a workaround as we are forced to use -Werror together with -Wbad-function-cast. An easy solution is to just remove the explizit cast to uint32_t.

```
static bool checkreturn pb_enc_bool(pb_ostream_t *stream, const pb_field_t *field, const void *src)
{
    uint32_t value = safe_read_bool(src);
    PB_UNUSED(field);
    return pb_encode_varint(stream, value);
}
```

Maybe you can add the workaround to the next maintenance release.

PetteriAimonen, 2019-10-16:
Heh, and the reason I added that cast was to avoid warnings from some weird compilers.

I made a change with ` ? 1 : 0;` now, that should at least make it explicit what values are expected to come out of it.

It will probably be a while until next maintenance release, but binaries will be available at http://jpa.kapsi.fi/jenkins/

cstender, 2019-10-17:
Thank you very much for the quick response!

closed 2019-12-20 by PetteriAimonen