nanopb/nanopb #442: Problem with "group" in .proto file.

tasilio, 2019-10-18:
An error occurs while compiling the .proto file below. If you remove the "group" from the file, everything goes smoothly.  Below .proto file is the error log.

Where could the problem be?

```
syntax = "proto2";

message Exch { 
	optional group Service = 1 { 
    	required bool request = 1; 
    	optional bool setparam = 2; 
	} 
	optional group DateTime = 2 { 
		optional sint32 year = 1; 
		optional sint32 month = 2; 
		optional sint32 day = 3; 
		optional sint32 hours = 4; 
		optional sint32 minutes = 5; 
		optional sint32 seconds = 6; 
	} 
	optional group Coord = 3 {  
		optional sint32 lat = 1; 
		optional sint32 latdec = 2; 
		optional sint32 lon = 3; 
		optional sint32 londec = 4; 
	} 
	optional group Prog = 4 { 
		optional sint32 num = 1; 
		optional bool astro = 2; 
		optional bool sunrise = 3; 
		optional bool advance = 4; 
		optional bool relay_state = 5; 
		optional sint32 hours = 6 ; 
		optional sint32 minutes = 7; 
		optional sint32 dailyrep = 8; 
	} 
} 
```

protoc --nanopb_out=. unitimer.proto
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1816, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1796, in main_plugin
  File "__main__protoc-gen-nanopb__.py", line 1642, in parse_file
  File "__main__protoc-gen-nanopb__.py", line 1135, in __init__
  File "__main__protoc-gen-nanopb__.py", line 1193, in parse
  File "__main__protoc-gen-nanopb__.py", line 939, in __init__
  File "__main__protoc-gen-nanopb__.py", line 412, in __init__
NotImplementedError: 10
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.

PetteriAimonen, 2019-10-18:
Groups are not implemented, as they were already deprecated in 2011 when first versions of nanopb were written.

It's listed in documentation http://jpa.kapsi.fi/nanopb/docs/index.html "Limitations" section.

tasilio, 2019-10-21:
Thank you for your prompt response! I did not immediately pay attention to this paragraph of the documentation.

closed 2019-10-21 by tasilio