nanopb/nanopb #443: Encoded size macro isn't calculated when using two-command method

toodv93, 2019-10-19:
Hi,
I have a related issue to #415.

It manifests itself that it works fine to generate with the one-command method:
```
FILE_NAMES=$(ls -1 *.proto | sed -e 's/\..*$//')

for f in $FILE_NAMES
do
  $PBPATH/protoc --plugin=$PBPATH/protoc-gen-nanopb --nanopb_out=.  $f.proto
done
```
Will the output be:
`#define Protobuf_size    462`

But when running it as a two commands generation:
```
FILE_NAMES=$(ls -1 *.proto | sed -e 's/\..*$//')

for f in $FILE_NAMES
do
  $PBPATH/protoc -o$f.pb $f.proto
  python $PBPATH/nanopb_generator.py --no-timestamp $f.pb
done
```
Will the output be something like:
`#define Protobuf_size    (11 + sizeof(union{char f2[No ...`

_Side note:_
The reason that I run the two-command way is that I want to take away timestamps from headers.

Best regards
Tomas

PetteriAimonen, 2019-10-20:
Hmm yeah, there probably isn't a way to get the dependency files parsed when running nanopb_generator.py directly. That should probably be added, though it will require that all the .pb files are generated first and given as arguments in one go.

Maybe `protoc --nanopb_out=--no-timestamp:. foo.proto` would work for you in the meantime? Protoc's option passing to plugins has some problems with spaces and special characters, but it should work for this.

toodv93, 2019-10-21:
Thanks Petteri,
That works fine.

PetteriAimonen, 2019-12-17:
Now in master branch `nanopb_generator.py` also parses all files given on command line as possible dependencies. It also will automatically compile any `.proto` files. Also, `--no-timestamp` is now the default setting.

So the invocation can be simplified to: `python $PBPATH/nanopb_generator.py *.proto`

closed 2019-12-20 by PetteriAimonen