nanopb/nanopb #445: Throwing EXC_BAD_ACCESS since 0.3.904

Idomo, 2019-10-24:
Hi, I've just upgraded a few pods that probably have this library as a dependency and upgraded it to `0.3.904`.
Since then, I've started getting the `EXC_BAD_ACCESS` error in xCode that freezing my app (basically it not usable anymore).
* With `0.3.901`(the previous version that it used) it seems to work fine.

The crash is in the file `pb_encode.c` around line 64:
[image]

This is the log of versions from my last `pod update`:
```
Using AppAuth (1.2.0)
Using BetterSegmentedControl (1.0)
Using Crashlytics (3.14.0)
Using CropView (0.1.6)
Using FBSDKCoreKit (5.2.3)
Using FBSDKLoginKit (5.2.3)
Using Fabric (1.10.2)
Installing Firebase 6.11.0 (was 6.9.0)
Installing FirebaseAnalytics 6.1.3 (was 6.1.2)
Installing FirebaseAuth 6.3.1 (was 6.2.3)
Using FirebaseAuthInterop (1.0.0)
Installing FirebaseCore 6.3.2 (was 6.3.0)
Installing FirebaseCoreDiagnostics 1.1.1 (was 1.1.0)
Using FirebaseCoreDiagnosticsInterop (1.0.0)
Installing FirebaseDatabase 6.1.1 (was 6.1.0)
Installing FirebaseInstanceID 4.2.6 (was 4.2.5)
Using FirebaseStorage (3.4.1)
Installing FirebaseUI 8.3.1 (was 8.1.0)
Using FontAwesomeSwift (1.9.2)
Using GTMAppAuth (1.0.0)
Installing GTMSessionFetcher 1.3.0 (was 1.2.2)
Installing Google-Mobile-Ads-SDK 7.51.0 (was 7.50.0)
Installing GoogleAppMeasurement 6.1.3 (was 6.1.2)
Installing GoogleDataTransport 3.0.1 (was 2.0.0)
Installing GoogleDataTransportCCTSupport 1.2.1 (was 1.1.0)
Installing GoogleSignIn 5.0.1 (was 5.0.0)
Using GoogleUtilities (6.3.1)
Using HGPlaceholders (0.4.0)
Using MBProgressHUD (1.1.0)
Installing SDWebImage 5.2.3 (was 5.2.2)
Using SideMenu (5.0.3)
Using SwiftDocumentScanner (0.1.2)
Using SwiftMessages (7.0.1)
Using Swime (3.0.6)
Using TweeTextField (1.6.0)
Using UITextField-Navigation (4.2)
Using UITextView+Placeholder (1.3.3)
Using leveldb-library (1.22)
Installing nanopb 0.3.904 (was 0.3.901)
```

PetteriAimonen, 2019-10-24:
Can you double-check that you have rebuilt the whole project after upgrade (clean and rebuild)?

This is probably related to this change that requires full rebuild: https://jpa.kapsi.fi/nanopb/docs/migration.html#fix-undefined-behavior-related-to-bool-fields

diegotl, 2019-10-24:
@PetteriAimonen That fixed for me. Thanks!

tylerwilson, 2019-10-24:
I had the same issue, though I was not directly including nanpb but it was included via Firebase. I tried a full rebuild and same crash, likely due to the fact that I am pulling down pre-built binaries. I added the following to my Podfile:

pod 'nanopb', '0.3.901'

and it appears to be working fine now.

rehannali, 2019-10-28:
I have the same issue and also tried rebuilt the whole project but since as mentioned 0.3.904, I have these crashes.

Idomo, 2019-10-28:
@rehannali Try to run “Clean build” after adding this to the `Podfile`.

rehannali, 2019-10-28:
@Idomo I tried and still having a problem with an unexpected crash. I downgrade to 0.3.904 and didn't crash for now.

xaphod, 2019-10-28:
+1: fresh pod install, then fresh clean build folder, complete rebuild, immediately crashes with error `<ERROR> Event decoder failed: varint overflow`. 

bt all:

```
(lldb) bt all
  thread #1, queue = 'com.apple.main-thread'
    frame #0: 0x0000000102272fc8 dyld`ImageLoaderMachO::findClosestSymbol(mach_header const*, void const*, void const**) + 276
    frame #1: 0x0000000102268810 dyld`dladdr + 220
    frame #2: 0x00000001c577d318 libdyld.dylib`dladdr + 164
    frame #3: 0x00000001d2d81f64 libswiftCore.dylib`replacementGetImageNameFromClass(objc_class*, char const**) + 60
    frame #4: 0x00000001c56b1e94 libobjc.A.dylib`class_getImageName + 36
    frame #5: 0x00000001c5c30454 Foundation`+[NSBundle bundleForClass:] + 248
    frame #6: 0x00000001012baf78 Photobooth`APMShouldSetAsDefaultScreenClass + 72
    frame #7: 0x00000001012bad44 Photobooth`-[APMScreenViewReporter viewControllerDidAppear:] + 236
    frame #8: 0x00000001012bc7c4 Photobooth`__44+[UIViewController(APMScreenClassName) load]_block_invoke_2 + 108
    frame #9: 0x00000001c92e06bc UIKitCore`-[UINavigationController viewDidAppear:] + 80
    frame #10: 0x00000001c937d2ec UIKitCore`-[UIViewController _setViewAppearState:isAnimating:] + 832
    frame #11: 0x00000001c937dcf8 UIKitCore`-[UIViewController _endAppearanceTransition:] + 228
    frame #12: 0x00000001c9297bac UIKitCore`__48-[UIPresentationController transitionDidFinish:]_block_invoke + 136
    frame #13: 0x00000001c998db98 UIKitCore`_runAfterCACommitDeferredBlocks + 316
    frame #14: 0x00000001c997d7c0 UIKitCore`_cleanUpAfterCAFlushAndRunDeferredBlocks + 232
    frame #15: 0x00000001c99ad594 UIKitCore`_afterCACommitHandler + 76
    frame #16: 0x00000001c58f3c48 CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 32
    frame #17: 0x00000001c58eeb34 CoreFoundation`__CFRunLoopDoObservers + 416
    frame #18: 0x00000001c58ef100 CoreFoundation`__CFRunLoopRun + 1308
    frame #19: 0x00000001c58ee8bc CoreFoundation`CFRunLoopRunSpecific + 464
    frame #20: 0x00000001cf75a328 GraphicsServices`GSEventRunModal + 104
    frame #21: 0x00000001c99846d4 UIKitCore`UIApplicationMain + 1936
    frame #22: 0x0000000100949f00 Photobooth`main(argc=3, argv=0x000000016f537700) at main.m:15:16
    frame #23: 0x00000001c5779460 libdyld.dylib`start + 4
  thread #3, queue = 'io.answers.EventQueue (QOS: BACKGROUND)'
    frame #0: 0x00000001c5914ab0 CoreFoundation`__CFStringCreateImmutableFunnel3 + 2292
    frame #1: 0x00000001c5914e28 CoreFoundation`CFStringCreateWithCharactersNoCopy + 56
    frame #2: 0x00000001c5d35698 Foundation`_NSNewStringByAppendingStrings + 660
    frame #3: 0x00000001c5c2d884 Foundation`-[NSString stringByAppendingString:] + 100
    frame #4: 0x00000001c5c30de0 Foundation`-[NSKeyValueUnnestedProperty _givenPropertiesBeingInitialized:getAffectingProperties:] + 364
    frame #5: 0x00000001c5c30e44 Foundation`-[NSKeyValueUnnestedProperty _givenPropertiesBeingInitialized:getAffectingProperties:] + 464
    frame #6: 0x00000001c5c30a50 Foundation`-[NSKeyValueUnnestedProperty _initWithContainerClass:key:propertiesBeingInitialized:] + 128
    frame #7: 0x00000001c5d24d4c Foundation`NSKeyValuePropertyForIsaAndKeyPathInner.llvm.13555634661636819103 + 280
    frame #8: 0x00000001c5c30e08 Foundation`-[NSKeyValueUnnestedProperty _givenPropertiesBeingInitialized:getAffectingProperties:] + 404
    frame #9: 0x00000001c5c30a50 Foundation`-[NSKeyValueUnnestedProperty _initWithContainerClass:key:propertiesBeingInitialized:] + 128
    frame #10: 0x00000001c5d24d4c Foundation`NSKeyValuePropertyForIsaAndKeyPathInner.llvm.13555634661636819103 + 280
    frame #11: 0x00000001c5d254d4 Foundation`NSKeyValuePropertyForIsaAndKeyPath + 156
    frame #12: 0x00000001c5d23de0 Foundation`_NSKeyValueCreateImplicitObservationInfo + 208
    frame #13: 0x00000001c5c441b8 Foundation`-[NSOperation init] + 356
    frame #14: 0x00000001010b6530 Photobooth`-[ANSCompressLogOperation initWithUncompressedLogURL:compressedLogURL:] + 88
    frame #15: 0x00000001010bc91c Photobooth`-[ANSAnswersController operationsForUncompressedLog:] + 360
    frame #16: 0x00000001010bc650 Photobooth`-[ANSAnswersController rotateAndSubmitCurrentLog] + 204
    frame #17: 0x00000001010bc314 Photobooth`__37-[ANSAnswersController enqueueEvent:]_block_invoke + 756
    frame #18: 0x00000001c5d3b61c Foundation`__NSBLOCKOPERATION_IS_CALLING_OUT_TO_A_BLOCK__ + 16
    frame #19: 0x00000001c5c453d8 Foundation`-[NSBlockOperation main] + 100
    frame #20: 0x00000001c5d3d8a4 Foundation`__NSOPERATION_IS_INVOKING_MAIN__ + 20
    frame #21: 0x00000001c5c45070 Foundation`-[NSOperation start] + 732
    frame #22: 0x00000001c5d3e29c Foundation`__NSOPERATIONQUEUE_IS_STARTING_AN_OPERATION__ + 20
    frame #23: 0x00000001c5d3dd68 Foundation`__NSOQSchedule_f + 180
    frame #24: 0x0000000109b81828 libdispatch.dylib`_dispatch_call_block_and_release + 24
    frame #25: 0x0000000109b82c04 libdispatch.dylib`_dispatch_client_callout + 16
    frame #26: 0x0000000109b85808 libdispatch.dylib`_dispatch_continuation_pop + 524
    frame #27: 0x0000000109b84d20 libdispatch.dylib`_dispatch_async_redirect_invoke + 776
    frame #28: 0x0000000109b94168 libdispatch.dylib`_dispatch_root_queue_drain + 376
    frame #29: 0x0000000109b94a90 libdispatch.dylib`_dispatch_worker_thread2 + 156
    frame #30: 0x00000001c5693f64 libsystem_pthread.dylib`_pthread_wqthread + 212
  thread #4, name = 'gputools.smt_poll.0x2825207e0'
[... cut ...] tuple_indices<> @ 0x000000016fdf2f58)::$_0>&, std::__1::__tuple_indices<>) at thread:342:5
    frame #5: 0x00000001077d7b1c Realm`void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, realm::_impl::ExternalCommitHelper::ExternalCommitHelper(realm::_impl::RealmCoordinator&)::$_0> >(__vp=0x0000000282736dc0) at thread:352:5
    frame #6: 0x00000001c56931ec libsystem_pthread.dylib`_pthread_start + 124
```

Idomo, 2019-11-18:
It seems to be fixed after upgrading to `nanopb 0.0.9011` & `Firebase 6.13.0` (maybe this related: https://github.com/firebase/firebase-ios-sdk/pull/3914/files).

Anyone else can approve this?

PetteriAimonen, 2019-12-20:
This sounds like some kind of mixup between version on the firebase side.

I guess it is related to nanopb not having a stable ABI between versions, so mixing headers (pb.h) from one version with code (pb_encode.c) from another version will not work correctly.

But there is nothing concrete I can do about this so I'm closing this for now.

closed 2019-12-20 by PetteriAimonen