nanopb/nanopb #446: Access via field tag?

selangh, 2019-10-28:
Hello,

is it possible to index a protobuf struct field via its tag number? As far as I know I cant "iterate" over the elements of a protobuf in C/C++ (ESP8266). So to match my existing signal list for reading data from a modbus device I was wondering if it would be possible to write values to a protobuf struct based on the field tag?

Something like
Protofile:
```
message TrovisData {
    required uint32 DeviceId = 5;
    required float Temperature = 242;
    required float Pump_State = 1043;
}
// as tag number I would use the modbus register numbers
```

C-Code:
```
TrovisData my_msg = TrovisData_init_zero;

tag_num = 5;
uint32 uint32_val = 12345;
kind_of_writefunc_uint32(&my_msg, tag_num, uint32_val);

tag_num = 242 or 1043;
float float_val = 1.34;
kind_of_writefunc_float(&my_msg, tag_num, float_val);
```

Is this possible?

PetteriAimonen, 2019-10-28:
It is possible, but the API is not very well documented. Also, the field iterator API had some quite large changes between 0.3.x series and the current master branch, which will eventually become 0.4.0. The new API should be easier to use.

The field iterator functions are defined in [`pb_common.h`](https://github.com/nanopb/nanopb/blob/master/pb_common.h). They provide access to a [`pb_field_iter_t`](https://github.com/nanopb/nanopb/blob/master/pb.h#L262) description of the fields.

Here's a simple example:

    pb_field_iter_t iter;
    pb_field_iter_begin(&iter, TrovisData_fields, &my_msg);
    if (pb_field_iter_find(&iter, tag_num))
    {
         // If we know beforehand the type of field, we can just write it.
         // Otherwise this should check iter.type to see whether it is int/float/bool etc.
         *(float*)iter.pData = 123.4;
    }

selangh, 2019-10-28:
Thx for the feedback. I try this later the day and give feedback. I dont have the ESP here at work - its my home project :-). Btw. the data type of the field I know beforehand. But maybe you can give me a short hint how to check the datatype with "iter.type" since iter seems not to have an element called type.

selangh, 2019-10-28:
I just implemented your suggestion and it works fine. Thanks a lot so far! Can I modify the "has_...." field through the tag number as well to be enable "optional" fields?

PetteriAimonen, 2019-10-29:
Sure, that's available through `*(bool*)iter.pSize = true`.

selangh, 2019-10-30:
Thanks a lot. Got it up and running :-).

selangh, 2020-01-02:
After updating to nanopb 0.4.0 this `*(bool*)iter.pSize = true`causes my ESP8266 to crash? Has something changed in using this field?

PetteriAimonen, 2020-01-02:
Hmm, shouldn't have changed between the version in master branch in October and the 0.4.0 release.

For debugging, try to print out value of `iter.pSize` and the expected value `&msg.has_field`.

selangh, 2020-01-02:
This is how to proto file looks like

```bash
syntax = "proto3";
package protobuf;
...
message TrovisRegisters {
    float Anlagenkennziffer = 1;
    float Firmware_Version = 2;
    float Hardware_Version = 3;
...
}
```

This is the c-code:
```c
...
  // Init protobufs and iterators
  protobuf_TrovisRegisters TrovisRegisters = protobuf_TrovisRegisters_init_zero;
  pb_field_iter_t TrovisRegisters_iter;
  pb_field_iter_begin(&TrovisRegisters_iter, protobuf_TrovisRegisters_fields, &TrovisRegisters);

...

      // iterate over all field to find the match between modbus-id and field tag
      if (pb_field_iter_find(&TrovisRegisters_iter, SignalPtr->ID)) {
        // write value
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " - before assignment of pData");
        *(float*)TrovisRegisters_iter.pData = SignalPtr->Value;
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " = " + String(*(float*) TrovisRegisters_iter.pData));

        // set as used
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " - iter.pSize before assigment = " + String(*(bool*) TrovisRegisters_iter.pSize));
        *(bool*)TrovisRegisters_iter.pSize = true;
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " - iter.pSize after assigment = " + String(*(bool*) TrovisRegisters_iter.pSize));
      } 
```
And this code leads to
```bash
...
DEBUG: Anlagenkennziffer - before assignment of pData
DEBUG: Anlagenkennziffer = 2.10

Exception (28):
epc1=0x40203419 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
```
Exception 28 means "bad pointer // access to invalid address". So i even can´t access this field reading.

PetteriAimonen, 2020-01-02:
I meant, print the value of the pointer. (something like `String((uint32_t)TrovisRegisters_iter.pSize)`)

That's probably 0 (NULL). That's because you are using `syntax = "proto3"` where there is no separate `has_field` for normal fields.

selangh, 2020-01-02:
You´re right, it´s zero indeed. But is has been proto3 all the time. Unfortunatly values assigned to my proto struct within the iter function by `*(float*)TrovisRegisters_iter.pData = SignalPtr->Value;` were not serialized anymore. In the old nanopb version they were serialized when setting `*(bool*)TrovisRegisters_iter.pSize = true;`.

The fields of the proto struct were only serialized when I assign them explicitly in my code like `TrovisRegisters.Anlagenkennziffer = 1234;`... but I need to assign them within this iter function.

PetteriAimonen, 2020-01-02:
Well, `pSize` has been NULL for proto3 field all along. And `required uint32 DeviceId = 5;` in the initial code from October suggests you have been using proto2 instead at some point.

If setting the values does not work, I'd suggest printing the value of the `pData` pointer and comparing it to the expected pointer (`&TrovisRegisters.Anlagenkennziffer`). You can also check whether the value in the struct changes as you would expect.

selangh, 2020-01-02:
Hui.. your right second time :-). I really thougt having used proto3 all the time. So I´ve done  following now based on your suggestions:

```c
      if (pb_field_iter_find(&TrovisRegisters_iter, SignalPtr->ID)) {
        // write value
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " value of pData before assignmnt " + String((uint32_t)TrovisRegisters_iter.pData));
        TrovisRegisters_iter.pData = &SignalPtr->Value; //SignalPtr->Value;
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " value of pData after assignment " + String((uint32_t)TrovisRegisters_iter.pData));
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " address of value to be assigned " + String((uint32_t) &SignalPtr->Value));
        Serial.println("DEBUG: " + String(SignalPtr->Name) + " float val pData now points to   " + String(*(float*) TrovisRegisters_iter.pData));

        Serial.println("DEBUG: Anlagenkennziffer pointer val from proto struct   " + String((uint32_t) &TrovisRegisters.Anlagenkennziffer));
        Serial.println("DEBUG: Anlagenkennziffer float val from proto struct     " + String((float) TrovisRegisters.Anlagenkennziffer));
        Serial.println("----------------------");
      }
```

```bash
Anlagenkennziffer = 2.10
DEBUG: Anlagenkennziffer value of pData before assignmnt 1073741264
DEBUG: Anlagenkennziffer value of pData after assignment 1073644952
DEBUG: Anlagenkennziffer address of value to be assigned 1073644952
DEBUG: Anlagenkennziffer float val pData now points to   2.10
DEBUG: Anlagenkennziffer pointer val from proto struct   1073741264
DEBUG: Anlagenkennziffer float val from proto struct     0.00
```

So it seems the pointer assignment to pData don´t affect the `&TrovisRegisters.Anlagenkennziffer`

... need to go out with the kids now. Will come back in some hours. Thanks for your help :-)

PetteriAimonen, 2020-01-02:
Well `TrovisRegisters_iter.pData = &SignalPtr->Value;` doesn't actually touch the struct at all, it just changes the pointer in the iterator. You want `*TrovisRegisters_iter.pData = SignalPtr->Value;` instead to change the value pointed by the pData pointer.

But from the debug info:

     DEBUG: Anlagenkennziffer value of pData before assignmnt 1073741264
     DEBUG: Anlagenkennziffer pointer val from proto struct   1073741264

we can infer that the pData correctly points to TrovisRegisters.Anlagenkennziffer.

selangh, 2020-01-02:
So now after all this debugging I am back at a point where I already was :-). That means the iter function looks like this:
```c
      if (pb_field_iter_find(&TrovisRegisters_iter, SignalPtr->ID)) {
        // write value
        *(float*) TrovisRegisters_iter.pData = SignalPtr->Value; //SignalPtr->Value;
      } else {
        Serial.println("Missing protobuf field-id for signal: " + String(SignalPtr->Name) + " with id: " + String(SignalPtr->ID));
      }
```
and it seems it does its job pretty goot because the serial output of

```c
  ...
  protobuf_TrovisRegisters TrovisRegisters = protobuf_TrovisRegisters_init_zero;
  protobuf_TrovisCoils TrovisCoils = protobuf_TrovisCoils_init_zero;
  protobuf_ClientState ClientState = protobuf_ClientState_init_zero;
  protobuf_TrovisClientTxData TrovisClientTxData = protobuf_TrovisClientTxData_init_zero;
  ...
  // 1. write client info into "client message"
  strcpy(ClientState.Client_IP ,       WiFi.localIP().toString().c_str());
  strcpy(ClientState.Client_Mac,       WiFi.macAddress().c_str());
  strcpy(ClientState.Client_Wifi_SSID, WiFi.SSID().c_str());
  strcpy(ClientState.Client_Version,   compile_date);
  ClientState.Client_Wifi_Signal     = WiFi.RSSI();
  ClientState.Client_UptimeDays      = float(millis())/86400000.0;
  ClientState.Client_FreeHeapBytes   = ESP.getFreeHeap();

  // 2. build up TrovisClientTxData message out of its child messages
  TrovisClientTxData.ClientState     = ClientState;
  TrovisClientTxData.TrovisCoils     = TrovisCoils;
  TrovisClientTxData.TrovisRegisters = TrovisRegisters;
  TrovisClientTxData.Trovis_Id       = TrovisControllerID;
  TrovisClientTxData.Trovis_Type     = TrovisErzeugnisnummer;

  // check if all assigned values were present
  Serial.println("Heizungspumpe_UP1    : " + String(TrovisClientTxData.TrovisCoils.Heizungspumpe_UP1));
  Serial.println("Speicherladepumpe_SLP: " + String(TrovisClientTxData.TrovisCoils.Speicherladepumpe_SLP));
  Serial.println("Zirkulationspumpe    : " + String(TrovisClientTxData.TrovisCoils.Zirkulationspumpe_ZP));
  Serial.println("Anlagenkennziffer    : " + String(TrovisClientTxData.TrovisRegisters.Anlagenkennziffer));
  Serial.println("Wifi name            : " + String(TrovisClientTxData.ClientState.Client_Wifi_SSID));
  Serial.println("TrovisId             : " + String(TrovisClientTxData.Trovis_Id));

  // Serialize protobuf and send:
  uint8_t pb_stream_buffer[1024] = {0};
  pb_ostream_t stream = pb_ostream_from_buffer(pb_stream_buffer, sizeof(pb_stream_buffer));

  if (!pb_encode_submessage(&stream, protobuf_TrovisRegisters_fields, &TrovisClientTxData.TrovisRegisters)) {
    Serial.println("failed to encode TrovisRegisters proto");
    return;
  } else {
    Serial.println("Encoded submessage TrovisRegisters.      Bytes written to stream buffer: " + String(stream.bytes_written));
  }

  if (!pb_encode_submessage(&stream, protobuf_ClientState_fields, &TrovisClientTxData.ClientState)) {
    Serial.println("failed to encode ClientState proto");
    return;
  } else {
    Serial.println("Encoded submessage ClientState.          Bytes written to stream buffer: " + String(stream.bytes_written));
  }

  if (!pb_encode_submessage(&stream, protobuf_TrovisCoils_fields, &TrovisClientTxData.TrovisCoils)) {
    Serial.println("failed to encode TrovisCoils proto");
    return;
  } else {
    Serial.println("Encoded submessage TrovisCoils.          Bytes written to stream buffer: " + String(stream.bytes_written));
  }

  if (!pb_encode(&stream, protobuf_TrovisClientTxData_fields, &TrovisClientTxData)) {
    Serial.println("failed to encode TrovisClientTxData proto");
    return;
[... 67 lines cut ...]
                TrovisDataSet = trovis_pb2.TrovisClientTxData()
                try:
                    TrovisDataSet.ParseFromString(data)
                except:
                    print("error parsing")

                pprint.pprint(TrovisDataSet)

            else:
                # no more data -- quit the loop
                print("no more data.")
                break
    finally:
        # Clean up the connection
        connection.close()

```
It says it receives this 280 bytes but deserialization fails
```bash
waiting for a connection
New connection from client ('10.20.30.18', 56884)
just received 280 bytes of data
error parsing
```
When trying this `TrovisDataSet.ParseFromString(data)` in debugger it tells me: `{DecodeError}Wrong wire type in tag.` :-( Now I am not sure if I do the serialization in c in a right way.

PetteriAimonen, 2020-01-03:
The way you are calling `pb_encode_submessage()` a couple of times before calling `pb_encode()` for the main message is not a very typical way of doing things.

Because you've already set `TrovisClientTxData.TrovisRegisters = TrovisRegisters;` into the main message, just one `pb_encode()` is enough and it will encode everything. But you'll need to set `TrovisClientTxData.has_TrovisRegisters = true;` first also (that's a change after the October version, #308).

(If, for some reason, you really wanted to call `pb_encode_submessage()` separately for each submessage, you'd have to have something like `pb_encode_tag(stream, PB_WT_STRING, TrovisClientTxData_TrovisRegisters_tag);` before each one.)

selangh, 2020-01-03:
Good morning and thanks a lot for your help. I´ve found this `pb_encode_submessage` in an example. But with your latest info (`TrovisClientTxData.has_TrovisRegisters = true;`) applied the whole message with its submessages is serialized and could be deserialized in Python again :-). There is only one problem left. Every field thats value is zero is not serialized. What do I need to do to get this 0-value fields serialized too?

selangh, 2020-01-03:
Ah.. I´ve just stumbled over this: https://developers.google.com/protocol-buffers/docs/proto3#default So this behaviour seems to be correct - values matching the default value of the field wont be serialized. So I will build a work arround using a special value to represent a zero value.

PetteriAimonen, 2020-01-03:
The normal way in proto3 is just to assume missing fields are zero, so don't care about how they are not present in the wire format.

selangh, 2020-01-03:
I use a single proto3 definition that matches different configuration sets of the Samson Trovis district heating controller. So there are fields defined that are used (and can be zero as valid value) and some that are not used (they are zero by default) depending on the Trovis config.
The simple assumption that zero means not existent don´t do the trick in this situation as long as I dont implement the logic what kind of config is actually running on the Trovis device twice (serverside and on my ESP). I thougt I could enable serialization of fields that are zero and should be transmitted as zero too. Due to the fact that this is not possible I have changed all bool datatypes to int32 ones and send 65535 as alternative value to a zero value. So my server just checks the data object for zeros and dismiss them, for 65535 and interpret this as zero value or it just uses the value from the deserialized proto struct.

closed 2020-01-03 by selangh