nanopb/nanopb #448: iOS App Store publishment fails: ITMS-90338 - Non-public API usage

shatk0vskiy, 2019-10-31:
[image]
Someone came across that?

PetteriAimonen, 2019-10-31:
Sounds like Apple uses nanopb internally, and the symbol names conflict.

Currently nanopb doesn't have a way to easily rename the symbols, but there are a few options you could explore:

- Stripping the private symbols after compilation. If you have nanopb linked in your own application, there is no need to expose those symbols in the final binary. https://stackoverflow.com/a/18226723 seems like a good start.

- Making a huge list of `#define`'s that would rename nanopb symbols at compile time. That could probably made with some kind of `grep -o 'pb_[a-z_]*' *.h | sed ...something ... | sort | uniq | xargs -n 1 printf '#define %s xyz_%s'` script, and then put into `pb_syshdr.h`.

closed 2019-10-31 by shatk0vskiy

khambir, 2019-11-19:
@shatk0vskiy may you help me, what was the solution for this issue?