nanopb/nanopb #449: Adding custom options without PB_FIELD_16B

elcojacobs, 2019-11-07:


elcojacobs, 2019-11-07:
I'm trying to find a workaround for the following issue:

I have a few custom field options and message options that we use for custom behavior when when parsing.

```

message BrewBloxMessageOptions {
  optional BrewBloxTypes.BlockType objtype = 3;
  repeated BrewBloxTypes.BlockType impl = 9 [ (nanopb).max_count = 5 ];
}

message BrewBloxFieldOptions {
  optional BrewBloxTypes.UnitType unit = 1;
  optional uint32 scale = 2;
  optional BrewBloxTypes.BlockType objtype = 3;
  optional bool hexed = 4;
  optional bool readonly = 5;
  optional bool logged = 6;
  optional bool hexstr = 7;
  optional bool driven = 8;
}

extend google.protobuf.FieldOptions { optional BrewBloxFieldOptions brewblox = 1011; }

extend google.protobuf.MessageOptions { optional BrewBloxMessageOptions brewblox_msg = 1011; }
```

With above, the compiler complains that 1011 does not fit in 8 bits and I should enable PB_FIELD_16BIT. I would rather not do that.
When I use a tag lower than 256, protoc complains that it is not an extension number.

When I use a separate options file, all options defined in the file are assumed to be nanopb options, but it does not provide an extension field.

Is there a way to make custom options work while keeping 8 bit fields?

PetteriAimonen, 2019-11-08:
Perhaps skipping the messages completely in nanopb generation would be ok for you? After all, nanopb doesn't really provide access to the extension fields anyway.

So try putting this in your foo.options file:

    BrewBloxMessageOptions skip_message:true
    BrewBloxFieldOptions skip_message:true
    brewblox type:FT_IGNORE
    brewblox_msg type:FT_IGNORE

PetteriAimonen, 2019-11-08:
Oh, and another possibility is if your options are all in a separate file (like `nanopb.proto` is), you can give `-x foo.proto` to avoid generating `#include` for it.

elcojacobs, 2019-11-08:
The options are indeed mostly intended for the python side, I'll try what I can do with skip and ignore, thanks!

elcojacobs, 2019-11-15:
I forgot to let you know, but these simple additions work fine:
``` 

message BrewBloxMessageOptions {
  option (nanopb_msgopt).skip_message = true;
  optional BrewBloxTypes.BlockType objtype = 3;
  repeated BrewBloxTypes.BlockType impl = 9 [ (nanopb).max_count = 5 ];
}

message BrewBloxFieldOptions {
  option (nanopb_msgopt).skip_message = true;

  optional BrewBloxTypes.UnitType unit = 1;
  optional uint32 scale = 2;
  optional BrewBloxTypes.BlockType objtype = 3;
  optional bool hexed = 4;
  optional bool readonly = 5;
  optional bool logged = 6;
  optional bool hexstr = 7;
  optional bool driven = 8;
}

extend google.protobuf.FieldOptions { optional BrewBloxFieldOptions brewblox = 50001 [(nanopb).type = FT_IGNORE ]; }

extend google.protobuf.MessageOptions { optional BrewBloxMessageOptions brewblox_msg = 50001 [ (nanopb).type = FT_IGNORE ]; }
```
Thanks for the support.

closed 2019-11-15 by elcojacobs