nanopb/nanopb #456: nanopb continues to compile if options have unmatched fields

wmmc88, 2019-11-23:
`Following patterns in <filename>.options did not match any fields: <list of field names>`

If you specify options on a certain field name, and the generator cannot find it, it will print the above message, however the compilation process continues and files are still produced. I believe the correct behaviour should be for the process to terminate after that error message. If someone were to rename a field in their .proto file, and forget to do the equivalent rename in their .optiosn file, they would still get a compiled output, but with incorrect options.

PetteriAimonen, 2019-11-24:
Well, generally that would result in a compile-time error when compiling the C files.

But it is a reasonable feature to have available via some `--error-on-unmatched` option to the generator. I probably won't be implementing it any time soon, but pull requests are welcome.

wmmc88, 2019-11-25:
I'm not sure how this would result in a compile time error?

PetteriAimonen, 2019-11-25:
That depends on the exact options used, but in many cases it will affect the field type, which would end up as an error in C.

PR #458 (2019-11-29): change generator to raise error on unmatched options and add cli option to disable
closes #456
closed 2019-11-29 by PetteriAimonen