nanopb/nanopb #457: pb_decode return status 1 but message was not decoded correcly

LTLinh2211, 2019-11-26:
I try to use pb_decode() to decode my input Byte Array in Arduino. The input byte array under the name Data[] is created from a HexString under the Name HexData[] by the function hexCharacterStringToBytes(Data, HexData);
`const int MaxByteArraySize = 200;
byte Data[MaxByteArraySize] = {0};
//char HexData[] = "0a7e0a17080010031a0973656e736f7269723120012a04434f32200a18080210021a0a73656e736f726361743120032a04636834200a17080410011a0973656e736f7265633120022a04434f20200a17080610011a0973656e736f7265633320022a044e4f32200a17080810011a0973656e736f7265633520022a04534f3220"; //ResponseGetAvailableChannels 
char HexData[] = "1289010A1908001801255D8FC23D280330AAF7F3EE053A04434F322040030A19080218032500000000280030AAF7F3EE053A046368342040030A1908041802250000803F280030AAF7F3EE053A04434F202040030A19080618022500000000280230AAF7F3EE053A044E4F322040030A19080818022500000000280130AAF7F3EE053A04534F322040031800"; //ResponseMeasurement 
bool status;
void setup(){
  Serial.begin(115200);
  hexCharacterStringToBytes(Data, HexData);
  dumpByteArray(Data, MaxByteArraySize);
  size_t message_length = sizeof(HexData)/2;
  ResponseBluetooth message = ResponseBluetooth_init_zero; 
  pb_istream_t stream = pb_istream_from_buffer(Data, message_length); 
  status = pb_decode(&stream, ResponseBluetooth_fields, &message);
  Serial.print("status: ");Serial.println(status);
  if (!status)
        {
            Serial.print("Decoding failed: "); Serial.print(PB_GET_ERROR(&stream)); return 1;
        }
  printMessageMeas(message);
}`
My ResponseBluetooth meassage has the following structure. Hier is a part of the generate C Code after running nanopb.exe.
`typedef struct _ResponseBluetooth {
    pb_size_t which_response;
    union {
        ResponseGetAvailableChannels channels;
        ResponseMeasurement measurement;
        ResponseBluetoothGetTimestamp timestamp;
    } response;
} ResponseBluetooth;`
when I tried to decode the first HexData[] //ResponseGetAvailableChannels , I got the right value of the structure ResponseGetAvailableChannels back. But when I tried to decode the second HexData[] //ResponseMeasurement the Status-Code of the the pb_decode() function is 1 but the variables in ResponseMeasurement structure either are equal zero or have strange values. The same Thing happend when I tried to run the Code in C with my Computer.
Does that means, that the function pb_decode() doesn't work correctly? I would greatly appreciate for any help

PetteriAimonen, 2019-11-26:
Did you check the value of `which_response`? Is it `ResponseBluetooth_measurement_tag`?

The message hex data looks ok when decoded with https://protogen.marcgravell.com/decode

Can you send the .proto file you are using?

LTLinh2211, 2019-11-28:
hi, thank you for answering. We find out that the problem was that I use the wrong version of pb and pb_decode file. When I use other pb_decode file it works.

closed 2019-12-16 by PetteriAimonen