nanopb/nanopb #459: Unable to build example fromdocs page

Omcsesz, 2019-11-30:
Hello team!

When I try to run the "Getting started" example of the following page:
https://jpa.kapsi.fi/nanopb/docs/index.html

I get an error in Atollic TRUEStudio:
```
..\Src\main.c: In function 'main':
..\Src\main.c:80:2: error: variable 'stream' has initializer but incomplete type
  pb_ostream_t stream= pb_ostream_from_buffer(buffer, sizeof(buffer));
  ^~~~~~~~~~~~
..\Src\main.c:80:23: warning: implicit declaration of function 'pb_ostream_from_buffer' [-Wimplicit-function-declaration]
  pb_ostream_t stream= pb_ostream_from_buffer(buffer, sizeof(buffer));
                       ^~~~~~~~~~~~~~~~~~~~~~
..\Src\main.c:80:15: error: storage size of 'stream' isn't known
  pb_ostream_t stream= pb_ostream_from_buffer(buffer, sizeof(buffer));
               ^~~~~~
..\Src\main.c:81:2: warning: implicit declaration of function 'pb_encode' [-Wimplicit-function-declaration]
  pb_encode(&stream, Example_fields, &mymessage);
  ^~~~~~~~~
..\Src\main.c:80:15: warning: unused variable 'stream' [-Wunused-variable]
  pb_ostream_t stream= pb_ostream_from_buffer(buffer, sizeof(buffer));
               ^~~~~~
```

Is it related to the example being incorrect, or is Atollic not set correctly?

PetteriAimonen, 2019-11-30:
The example on that page does not show the `#include`s that are needed. I updated it now to have them also.

For a complete example, see here:
https://github.com/nanopb/nanopb/blob/master/examples/simple/simple.c

Omcsesz, 2019-11-30:
It fixed it, thanks!
One small thing though: when I use max_size option in the .proto file, i get an error:
`Command.proto:7:31: Option "(nanopb)" unknown.`

My .proto file:
```
syntax = "proto2";

message Command {
    required string name = 1 [(nanopb).max_size = 40];
}
```

My option file (it is in the dame folder, and has the same name):
`(nanopb).max_size 		max_size:40`

What do I do wrong?

PetteriAimonen, 2019-11-30:
Again missing includes, namely `import "nanopb.proto";`

Omcsesz, 2019-11-30:
I added that as well:

```
syntax = "proto2";

import "proto/nanopb.proto";

message Command {
    required string name = 1 [(nanopb).max_size = 40];
}
```
But now I get another error:
```google/protobuf/descriptor.proto: File not found.
proto/nanopb.proto: Import "google/protobuf/descriptor.proto" was not found or had errors.
proto/nanopb.proto:137:8: "google.protobuf.FileOptions" is not defined.
proto/nanopb.proto:141:8: "google.protobuf.MessageOptions" is not defined.
proto/nanopb.proto:145:8: "google.protobuf.EnumOptions" is not defined.
proto/nanopb.proto:149:8: "google.protobuf.FieldOptions" is not defined.
Command.proto: Import "proto/nanopb.proto" was not found or had errors.
```

PetteriAimonen, 2019-11-30:
Yeah, you'll probably need to fix your include paths, something like `-I path/to/nanopb/generator/proto` should work.

Omcsesz, 2019-11-30:
Like so?

```
protoc --nanopb_out=. Command.proto -I.\nanopb\generator\proto\
pause
```

Omcsesz, 2019-11-30:
Update: with the following cmd commands:
```
protoc --nanopb_out=. -proto-path=d:\github\nanopb-0.3.9.4-windows-x86\nanopb-0.3.9.4-windows-x86\generator\ Command.proto 
pause
```
I got the following error:
```
Unknown flag: -p
```

PetteriAimonen, 2019-12-01:
Yeah, protoc is a bit picky about include paths. Just read the error message and add `-I.` or something.

closed 2019-12-16 by PetteriAimonen