nanopb/nanopb #461: nanopb 0.3.9.4 pip package is missing some generated files.

theunkn0wn1, 2019-12-13:
The `nanopb==0.3.9.4` package on pypi appears to be missing some generated files, resulting in build failures.

## Steps to reproduce
```
PS C:\Users\orion> docker run --rm -it python:3.8 /bin/bash
root@4e90c588bebe:/# apt update && apt install -y protobuf-compiler
Get:1 http://security.debian.org/debian-security buster/updates InRelease [65.4 kB]
Get:2 http://deb.debian.org/debian buster InRelease [122 kB]
Get:3 http://deb.debian.org/debian buster-updates InRelease [49.3 kB]
Get:4 http://security.debian.org/debian-security buster/updates/main amd64 Packages [162 kB]
Get:5 http://deb.debian.org/debian buster/main amd64 Packages [7908 kB]
Get:6 http://deb.debian.org/debian buster-updates/main amd64 Packages [5792 B]
Fetched 8312 kB in 2s (5323 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libprotobuf-dev libprotobuf-lite17 libprotobuf17 libprotoc17
The following NEW packages will be installed:
  libprotobuf-dev libprotobuf-lite17 libprotobuf17 libprotoc17 protobuf-compiler
0 upgraded, 5 newly installed, 0 to remove and 2 not upgraded.
Need to get 2840 kB of archives.
After this operation, 16.6 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian buster/main amd64 libprotobuf17 amd64 3.6.1.3-2 [814 kB]
Get:2 http://deb.debian.org/debian buster/main amd64 libprotobuf-lite17 amd64 3.6.1.3-2 [167 kB]
Get:3 http://deb.debian.org/debian buster/main amd64 libprotobuf-dev amd64 3.6.1.3-2 [1130 kB]
Get:4 http://deb.debian.org/debian buster/main amd64 libprotoc17 amd64 3.6.1.3-2 [665 kB]
Get:5 http://deb.debian.org/debian buster/main amd64 protobuf-compiler amd64 3.6.1.3-2 [64.9 kB]
Fetched 2840 kB in 0s (14.0 MB/s)
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package libprotobuf17:amd64.
(Reading database ... 24526 files and directories currently installed.)
Preparing to unpack .../libprotobuf17_3.6.1.3-2_amd64.deb ...
Unpacking libprotobuf17:amd64 (3.6.1.3-2) ...
Selecting previously unselected package libprotobuf-lite17:amd64.
Preparing to unpack .../libprotobuf-lite17_3.6.1.3-2_amd64.deb ...
Unpacking libprotobuf-lite17:amd64 (3.6.1.3-2) ...
Selecting previously unselected package libprotobuf-dev:amd64.
Preparing to unpack .../libprotobuf-dev_3.6.1.3-2_amd64.deb ...
Unpacking libprotobuf-dev:amd64 (3.6.1.3-2) ...
Selecting previously unselected package libprotoc17:amd64.
Preparing to unpack .../libprotoc17_3.6.1.3-2_amd64.deb ...
Unpacking libprotoc17:amd64 (3.6.1.3-2) ...
Selecting previously unselected package protobuf-compiler.
Preparing to unpack .../protobuf-compiler_3.6.1.3-2_amd64.deb ...
Unpacking protobuf-compiler (3.6.1.3-2) ...
Setting up libprotobuf-lite17:amd64 (3.6.1.3-2) ...
Setting up libprotobuf17:amd64 (3.6.1.3-2) ...
Setting up libprotoc17:amd64 (3.6.1.3-2) ...
Setting up protobuf-compiler (3.6.1.3-2) ...
Setting up libprotobuf-dev:amd64 (3.6.1.3-2) ...
Processing triggers for libc-bin (2.28-10) ...
root@4e90c588bebe:/# pip install nanopb
Collecting nanopb
  Downloading https://files.pythonhosted.org/packages/bd/4f/cf9e9da38fc57786d5edcd7a5a9f2aa31fd1204aa7230b2589c1bbd3faea/nanopb-0.3.9.4-py2.py3-none-any.whl
Collecting protobuf>=3.6
  Downloading https://files.pythonhosted.org/packages/4e/26/1517e42a81de4f28ed0f3cdadb628c1b72f3a28f38323a05e251f5df0a29/protobuf-3.11.1-py2.py3-none-any.whl (434kB)
     |████████████████████████████████| 440kB 12.8MB/s
Collecting six>=1.9
  Downloading https://files.pythonhosted.org/packages/65/26/32b8464df2a97e6dd1b656ed26b2c194606c16fe163c695a992b36c11cdf/six-1.13.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools in /usr/local/lib/python3.8/site-packages (from protobuf>=3.6->nanopb) (41.6.0)
Installing collected packages: six, protobuf, nanopb
Successfully installed nanopb-0.3.9.4 protobuf-3.11.1 six-1.13.0
root@4e90c588bebe:/# mkdir project
root@4e90c588bebe:/# cd project
root@4e90c588bebe:/project# nanopb_generator ^C
root@4e90c588bebe:/project# protoc --nanopb_out=. empty.proto

         ********************************************************************
         *** Failed to import the protocol definitions for generator.     ***
         *** You have to run 'make' in the nanopb/generator/proto folder. ***
         ********************************************************************

Traceback (most recent call last):
  File "/usr/local/bin/protoc-gen-nanopb", line 5, in <module>
    from nanopb.generator.nanopb_generator import main_plugin
  File "/usr/local/lib/python3.8/site-packages/nanopb/generator/nanopb_generator.py", line 37, in <module>
    from .proto import nanopb_pb2, plugin_pb2
ImportError: cannot import name 'nanopb_pb2' from 'nanopb.generator.proto' (/usr/local/lib/python3.8/site-packages/nanopb/generator/proto/__init__.py)
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

PetteriAimonen, 2019-12-14:
Should be fixed now in 0.3.9.4.post3.

I backported the same `__init__.py` trick that autogenerates the files on 0.4.0.

This will still probably cause some issues in two cases:
1) If user doesn't have `protoc` installed. But that won't be very useful anyway. I guess there is no `pip` package that would include the compiler, even though IMO it would make sense that `protoc` would be included with the `protobuf` package considering the versions have to match.

2) If the `nanopb` package is installed as root, the first time it must be also run as root to be able to generate the files.

theunkn0wn1, 2019-12-14:
There actually is a python package to provide the *python* protoc generator, [`grpcio-tools`](https://pypi.org/project/grpcio-tools/).

It takes a slight adjustment to your auto-generation trick in `generator/proto/__init__.py` to invoke this generator instead of a system-installed protoc.

Did some initial testing, and using `grpcio.tools.protoc` does support the nanopb plugin (when installed as a python package), and produces valid C sources.

the effected commit becomes 
```py

'''This file automatically rebuilds the proto definitions for Python.'''

import os
import os.path
import sys
from grpc.tools import command  # from grpcio-tools
dirname = os.path.dirname(__file__)
protosrc = os.path.join(dirname, "nanopb.proto")
protodst = os.path.join(dirname, "nanopb_pb2.py")

if os.path.isfile(protosrc):
    src_date = os.path.getmtime(protosrc)
    if not os.path.isfile(protodst) or os.path.getmtime(protodst) < src_date:
        command.build_package_protos(dirname)

```
Installing the `protoc` compiler ahead-of-time is not required using this route.

PetteriAimonen, 2019-12-15:
Wow, this is great to know. I think it might make sense to add a mode for nanopb_generator that can take .proto files directly and transparently pass them through grpc.tool.protoc.

theunkn0wn1, 2019-12-15:
Thanks for releasing a fixed version, much appreciated.

I agree, if its not too much work using the python protoc distribution will make using this library a bit simpler. 

Im going to go ahead and close this ticket as resolved in release `0.3.9.4.post3`

closed 2019-12-15 by theunkn0wn1