nanopb/nanopb #466: Decode string with callback

hamid-rostami, 2019-12-17:
Hi,
I'm trying to encode and decode string with callback. Encode callback works fine and output stream tested with simple python script, but `pb_decode` function return false.
Here is test program, try to encode `lucky` text with `data_enc_cb` function, then write encoded strem to file (for test with python script) and then try to decode back with `data_dec_cb` function.

```c
bool data_enc_cb(pb_ostream_t *stream, const pb_field_t *field,
                 void *const *arg) {
  uint8_t buf[] = {"lucky"};
  bool ret;

  printf("Encoding data\n");

  if (!pb_encode_tag_for_field(stream, field))
    return false;

  return pb_encode_string(stream, buf, strlen(buf));
}

bool data_dec_cb(pb_istream_t *stream, const pb_field_t *field, void **arg) {
  uint8_t buf[10];

  printf("Decoding data\n");
  if (!pb_decode(stream, field, buf)) {
    printf("Failed\n");
    return false;
  }

  printf("decde data success\n");
  return true;
}

int main(int argc, char **argv) {
  Cmd cmd = {};
  Cmd cmd2 = {};
  uint8_t obuffer[50];
  pb_ostream_t ostream;
  pb_istream_t istream;
  bool ret;
  FILE *fp = fopen("enc.bin", "w");

  /* Encode */
  cmd.data.funcs.encode = &data_enc_cb;

  ostream = pb_ostream_from_buffer(obuffer, sizeof(obuffer));
  ret = pb_encode(&ostream, Cmd_fields, &cmd);

  if (!ret) {
    printf("Encode Failed\n");
    return -1;
  }
  printf("Encoding success, bytes: %d\n", ostream.bytes_written);
  fwrite(obuffer, sizeof(obuffer[0]), ostream.bytes_written, fp);

  /* Decode */
  cmd2.data.funcs.decode = &data_dec_cb;
  istream = pb_istream_from_buffer(obuffer, ostream.bytes_written);
  ret = pb_decode(&istream, Cmd_fields, &cmd2);
  if (!ret) {
    printf("Decode Failed\n");
    return -1;
  }
  printf("Decode success\n");

  return 0;
}
```

`proto` file:
```
syntax = "proto3";

message Cmd {
  string data = 1;
}

```

Output of above program:
```
Encoding data
Encoding success, bytes: 7
Decoding data
Failed
Decode Failed
```

python test script:
```python
import sample_pb2

with open("enc.bin") as f:
    data = f.read().encode("ASCII")

cmd = sample_pb2.Cmd()
cmd.ParseFromString(data)

assert cmd.data == "lucky"
print("Success")
```

PetteriAimonen, 2019-12-17:
if (!pb_decode(stream, field, buf)) {

`pb_decode` works only for messages & submessages, but not for strings.

Use `pb_read` instead, length of the string is available in `stream->bytes_left`.

See e.g. https://github.com/nanopb/nanopb/blob/master/tests/callbacks/decode_callbacks.c#L10

hamid-rostami, 2019-12-17:
Thank you, It's very helpful.
But one more question, `stream->bytes_left` is exactly length of encoded string? (`data` field in this example) or depend on other fields. For example if I encode some fields before or/and after `data`, `stream->bytes_left` still give correct size of `data`?

PetteriAimonen, 2019-12-17:
Decoding callbacks receive a modified stream that contains only single field: https://jpa.kapsi.fi/nanopb/docs/concepts.html#decoding-callbacks

closed 2019-12-18 by hamid-rostami