nanopb/nanopb #470: ImportError: attempted relative import with no known parent package

aep, 2019-12-26:
according to the README, the binary release is supposed to be straight forward

```
generator-bin/protoc --nanopb_out=. myprotocol.proto
```

there is no generator-bin, so i'm assuming its   ./generator/protoc   ,  but that doesnt work

```
aep@stark: ~/Downloads/nanopb ./generator/protoc 

         ********************************************************************
         *** Failed to import the protocol definitions for generator.     ***
         *** You have to run 'make' in the nanopb/generator/proto folder. ***
         ********************************************************************
    
Traceback (most recent call last):
  File "./generator/protoc", line 4, in <module>
    from nanopb_generator import invoke_protoc
  File "/home/aep/Downloads/nanopb/generator/nanopb_generator.py", line 42, in <module>
    from .proto import nanopb_pb2
ImportError: attempted relative import with no known parent package

```

PetteriAimonen, 2019-12-26:
Hmm, if there is no `generator-bin`, it sounds like you have the plain source package instead of the linux binary package.

But even then, it should work the way you call it if you happen to already have either binary protoc or grpcio-tools Python package installed. The error you received seems to occur only with Python 3. I have fixed it now in revision 96670da.

aep, 2019-12-27:
i dunno what i did with the binary, but your latest commit fixes it with from-source builds, thanks

PetteriAimonen, 2019-12-31:
I'll reopen this until fix is released, to make it easier to find for others hitting the same problem.

closed 2020-02-02 by PetteriAimonen