nanopb/nanopb #471: python-protobuf required for OS X

alexgoldstone, 2019-12-27:
I have been using the previous version of nanopb without issue via Windows but 0.4.0 will just not work... I'm struggling to get any useful output other than it doesn't do anything so haven't raised this as an issue yet.

For now I thought it would be easier to just use the 0.4.0 OS X binaries but running the following command in the OS X terminal:
```
~/Downloads/nanopb-0.4.0-macosx-x86/generator/protoc --nanopb_out=. test.proto
```

Produces:
```
*************************************************************
*** Could not import the Google protobuf Python libraries ***
*** Try installing package 'python-protobuf' or similar.  ***
*************************************************************
```

I tried `pip install protobuf` which didn't help. Any suggestions?

closed 2019-12-27 by alexgoldstone

lmapii, 2019-12-27:
It might help to add `pip install protobuf --upgrade` to the instructions.

PetteriAimonen, 2019-12-28:
Yeah, I guess the installation part of docs should be broken to entirely separate page and explain in more detail the simple way (`pip install nanopb protobuf grpcio-tools`) and the alternate ways.

theunkn0wn1, 2019-12-30:
as of 0.4.0 you can do `pip install nanopb[grpcio-tools]` which will install the optional dependency.