nanopb/nanopb #472: submsg_callback fails to compile in Atmel Studio

alexgoldstone, 2019-12-27:
I have updated a codebase to use nanopb 0.4.0 in order to use the submsg_callback option.

I am using Atmel Studio 7 with an Atmel micro.

I have simplified my proto file to the following for testing:
```
message FirmwareUpdateCommand {
	string version = 1;
	uint32 size = 2;
}

message CommandWrapperMessage {
	option (nanopb_msgopt).submsg_callback = true;

    oneof payload {
		FirmwareUpdateCommand firmware_update = 1;
    }
}
```

If my proto file contains:
```
option (nanopb_msgopt).submsg_callback = false;
```

Then everything builds ok but I obviously cannot use submsg callbacks.

If I try to use:
```
option (nanopb_msgopt).submsg_callback = true;
```

Then I get a series of compilation errors:
[image]

With the first error pointing to:
[image]

I note that you have had a previous embedded systems issue and added a series of tests but it is possible these tests to not cover the use of submsg_callback being set to true.

PetteriAimonen, 2019-12-28:
The `oneof_callback` testcase covers that and is run on AVR also. But actually the particular combination of fields in your message caused a compilation error on any platform.

It's just due to a missing `#define` for the descriptor size logic. I've fixed this in newest git master.

In 0.4.0, you can also work around the bug by defining:

    	option (nanopb_msgopt).descriptorsize = DS_2;

alexgoldstone, 2019-12-29:
I haven't tested running on actual hardware yet but I have now confirmed that the master branch with this change now compiles in Atmel Studio which should close-out this issue.

Thank you.

PetteriAimonen, 2019-12-29:
Great to hear!

I'll keep this issue open until 0.4.1 is released, to make it easier to find if someone else hits it.

closed 2020-02-02 by PetteriAimonen