nanopb/nanopb #477: Is there a way to not omit a field whose value is zero?

andrew-mclachlan, 2020-01-07:
Hi there -

Is there a way to include all values in the encoded protobuf even when their value is zero?

I can see from `https://github.com/nanopb/nanopb/blob/master/pb_encode.c` that -- In proto3, all fields are optional and are only encoded if their value is "non-zero".

I'd like to be have zero values encoded. Is that possible?

PetteriAimonen, 2020-01-07:
Not encoding zero values is the standard encoding for proto3 syntax, as followed by Google's own protobuf libraries and most others.

If you really want to, you can force the individual fields to treated as proto2 fields by setting `(nanopb).proto3 = false`. They'll then have separate `has_` fields that you must set to true to have them encoded. The binary representation will be compatible with decoders using proto3 mode.

However, it would be interesting to know why you want this?

DrinkDeuchars, 2020-01-07:
As my Math teacher drummed into me all those years ago, "Zero is not nothing".  So I am actually interested to know why zero is not encoded?

PetteriAimonen, 2020-01-07:
@DrinkDeuchars You can find some of Google's reasoning in the release notes for proto3: https://github.com/protocolbuffers/protobuf/releases/tag/v3.0.0

Basically it boils down to simplicity in implementation between different languages, and better compatibility between different revisions of a .proto file. I'm not a huge fan of proto3 syntax myself, but I guess in some cases it is better than proto2.

mrtzcspr, 2020-01-08:
One way that is recommend and works quite well is to use a so called "wrapper" message.  
The trick is that proto3 still has the means to detect the presence of a message. So you wrap your scalar type in a minimal reusable message. If the message is not present the value is actually `not set`, if the message is present but the value is `0` then you know its really `0` .

Example Protofile:
```protobuf

message MyLogMessage{
  UInt32Value voltage_a = 1;
  UInt32Value voltage_b = 2;
  Int32Value temperature_outside = 3;
  Int32Value temperature_inside = 4;
}

/**************************************************************************************************
* Wrapper messages for scalar data types
* Used to distinguish between a missing field and the actual value
* see: 
* - https://developers.google.com/protocol-buffers/docs/reference/google.protobuf#google.protobuf.UInt32Value
* - https://github.com/protocolbuffers/protobuf/blob/master/src/google/protobuf/wrappers.proto
* for use with nanopb 0.3.x a workarround(https://github.com/nanopb/nanopb/issues/308) is needed to generate a has_*-field for this message
* by setting the option "proto3:false" for this message.
**************************************************************************************************/
message Int32Value {
  int32 value = 1;
}
message UInt32Value {
  uint32 value = 1;
}
```

then in your code you can do something like
```c
void create_log_message(void)
{
  // init message
  MyLogMessage msg = MyLogMessage_init_zero;

  // set voltage_a
  msg.has_voltage_a = true;
  msg.voltage_a = 100;

  // do something with your messsage
}

void receive_log_message(MyLogMessage *msg_p)
{
  if (msg_p->has_voltage_a)
  {
    // voltage_a field is set
    if (msg_p->voltage_a == 0)
    {
      // voltage_a is really 0
    }
  }
  else
  {
    // voltage_a field is not set
  }
}

```

andrew-mclachlan, 2020-01-08:
Thank you for all your amazing replies!

We're opting to use `proto2` and specifying `required` to ensure that we receive zeros.

If there's a chance of getting `required` or similar added to `proto3` so that fields can be kept (i.e. we keep zeros), then that would be great!!

PetteriAimonen, 2020-01-09:
@andrew-mclachlan Unfortunately there is little I can do about that, as the `.proto` files have to be compatible across all protobuf libraries. I think the main concerns about proto3 have been mentioned in Google's protobuf mailing list quite a few times over the years, and the basic reply is that proto2 is here to stay, and one can use it if required/optional division seems preferrable for an application.

Likewise, as far as I'm concerned, nanopb will keep proto2 support for ever.

andrew-mclachlan, 2020-01-09:
That's grand. Thank you for your reply.
Please feel free to close out this issue as Question Answered. Thanks again!

closed 2020-01-09 by andrew-mclachlan