nanopb/nanopb #478: Tests appear to be failing on 0.4.0 (and HEAD)

bolind, 2020-01-08:
When doing the following steps:

- git clone https://github.com/nanopb/nanopb.git
- cd nanopb
- cmake .
- make
- cd generator/proto
- make
- cd ../../tests
- scons

The tests fail somewhere in nanopb_generator.py:1200:

`Traceback (most recent call last):
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 2017, in <module>
    main_plugin()
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 2002, in main_plugin
    results = process_file(filename, fdesc, options, other_files)
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 1856, in process_file
    headerdata = ''.join(f.generate_header(includes, headerbasename, options))
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 1513, in generate_header
    yield msg.fields_declaration(self.dependencies) + '\n'
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 1092, in fields_declaration
    defval = self.default_value(dependencies)
  File "/home/bo.lind/nanopb/generator/nanopb_generator.py", line 1202, in default_value
    field.ClearField('default_value')
TypeError: field name must be a string`

This in on Ubuntu 18.04.3 LTS, x86_64.

PetteriAimonen, 2020-01-08:
Could you post the output of these two commands:

    python -c 'import google.protobuf; print(google.protobuf.__version__)'
    python -c 'import google.protobuf; print(google.protobuf.__file__)'

The 3.0.0 version that comes from apt-get in 18.04 is probably too old. I guess there should be a warning/error when trying to use with a too old version.

You can try installing python-protobuf 3.6 or newer using e.g. `pip install protobuf`.

bolind, 2020-01-08:
`bo.lind@redacted:~$ python -c 'import google.protobuf; print(google.protobuf.__version__)'
3.0.0
bo.lind@redacted:~$ python -c 'import google.protobuf; print(google.protobuf.__file__)'
/usr/lib/python2.7/dist-packages/google/protobuf/__init__.pyc`

Thanks a bunch, I'll see if I can scare up a newer machine!

bolind, 2020-01-09:
Am I correct in concluding that the issue persists on Ubuntu 20.4 (beta) and protoc 3.6, as shown by the following Dockerfile?

`FROM ubuntu:focal
MAINTAINER Bo Langgaard Lind <bo.langgaard.lind@gmail.com>

ENV TZ=Europe/Copenhagen

RUN apt -y update
RUN apt -y upgrade
RUN apt -y dist-upgrade
RUN apt -y autoremove
RUN apt -y install --fix-missing
RUN apt -y install apt-utils

RUN apt -y install protobuf-compiler
RUN apt -y install git
RUN apt -y install scons
RUN apt -y install cmake
RUN apt -y install python
RUN apt -y install python-protobuf
RUN apt -y install g++

RUN git clone https://github.com/nanopb/nanopb.git

RUN echo $(python -c 'import google.protobuf; print(google.protobuf.__version__)')
RUN echo $(python -c 'import google.protobuf; print(google.protobuf.__file__)')

RUN cd nanopb && cmake . && make
RUN cd nanopb/generator/proto && make
RUN cd nanopb/tests && scons`

PetteriAimonen, 2020-01-09:
No, shouldn't occur with such new versions, seems like this is something else after all.

I was now able to reproduce the problem with the dockerfile and I'm looking into it.

bolind, 2020-01-09:
OK, awesome, thanks for your speedy reply!

If you want to check for passing tests on Ubuntu 18.04 as well, I think the Dockerfile is sufficiently generic that you can just replace "focal" with "bionic" and test it out (same goes for 19.10 eoan.)

PetteriAimonen, 2020-01-09:
Aha, looks like it is a bug in python-protobuf 3.6: https://github.com/protocolbuffers/protobuf/issues/3658

This was already seen in #379 and a workaround was added but all cases weren't fixed.

Now it seems that actually the tests run ok on 18.04 also. `extra/FindNanopb.cmake` uses the new `--plugin_opt=` syntax introduced in protoc 3.6, so that won't work though.

Looks like I've had a gap in my test platform coverage. I'll add docker files for 18.04 and 20.4 to the regular test runs.

bolind, 2020-01-09:
This is awesome, thanks! And happy the Dockerfiles are now in.

closed 2020-02-02 by PetteriAimonen