nanopb/nanopb #480: Problem with partial decoding approach when migrating to 0.4.0

toodv93, 2020-01-13:
Facing a problem when testing to migrate from 0.3.9.4 to 0.4.0 due to that a used "feature" have stopped working.

Scenario - Having a generic message with a header message:

```
 message GenericMessage
 {
   Header header = 1; // Always first in message
 
   oneof msg
   {
     ...
     ...
     ...
   }
 }
```

The Header is used for routing message in the system. 
Basically using pattern discussed in article https://groups.google.com/forum/#!topic/nanopb/t2vBa_L5xTQ

This works fine in 0.3.9.4 but seams to be broken in 0.4.0. That is the decoder reports no error when only decoding the header, but the decoded values are wrong. 
Can it be due to the introduction of the has_*-parameter for sub-messages?

BR
Tomas

PetteriAimonen, 2020-01-14:
Hmm, I don't think there are any changes that should affect this.

Would you be able to provide a runnable example that reproduces the problem?

toodv93, 2020-01-14:
Thanks Petteri for your feedback.

I had a closer look today and the problem was that i wrote with pb_decode to a struct of Header-type when it should have been a struct that had the Header wrapped in a message.

It didn't cause any problems in 0.3.9.4, but now it became visible due to the has-field.

So to conclude: No bug in nanoPB.

Sorry for false alarm.

Best regards
Tomas

PetteriAimonen, 2020-01-14:
Good to hear you got it working :)

closed 2020-01-14 by PetteriAimonen