nanopb/nanopb #481: CMake support of NANOPB_GENERATE_CPP function is broken or incomplete

ArtemPisarenko, 2020-01-15:
CMake projects using nanopb should use `nanopb_generate_cpp` function as shown in cmake examples. But it's being provided only by `extra/FindNanopb.cmake` *cmake package module*, which available only in source tree and "exclusive" binary distributions at download page. Although, there are also *cmake package configuration* available, being generated from nanopb build and install via CMake, it doesn't provide any similar feature.

This limits usage in two possible deployment scenarios:
- nanopb source code isn't available during user application build. One need to retreive it and append [CMAKE_MODULE_PATH](https://cmake.org/cmake/help/latest/variable/CMAKE_MODULE_PATH.html) with path to extracted directory (which can be done in cmake command line, of course, and not as shown in example provided)
- nanopb already installed in system (linux). Recommended `find_package(Nanopb ...)` instruction from example causes CMake to find installed *cmake package configuration* first and [prefer to use it](https://cmake.org/cmake/help/latest/command/find_package.html#id2) , regardless of whether user supplied CMake with path to *module* location or not, thus causing build fail (not only because of missing function). This can be avoided by explicit `find_package(Nanopb MODULE ...)`.

Most simple solution would be to add installation of`FindNanopb.cmake` to standard cmake modules directory (and also core sources somewhere to be retreived from) and correct documentation/examples describing two different use cases:
- `find_package(Nanopb MODULE ...)` + available function and integration to user cmake file (as already shown in example but with CMAKE_MODULE_PATH required only when nanopb isn't installed)
- `find_package(Nanopb CONFIG ...)` + integration of imported targets to user cmake file and alternative invokation of generator

PetteriAimonen, 2020-01-16:
I'd like to point out that nanopb doesn't have a stable binary ABI, and is not designed to be used as a shared library. I'm not very much fan of how the cmake rules build a shared library, but I guess it can work as long as the library is only bundled with one application.

ArtemPisarenko, 2020-01-16:
Actually it doesn't have to be specifically *shared* library when installed to system. But I see that *static* library build is already supported. Or did you mean that it shouldn't be any library at all (due to some cmake issues) ? Anyway, if you decide to not support nanopb system install, then it should be done explicitly because now it can cause build failure I described in 2nd deployment scenario (where user wishes to use nanopb source tree, but doesn't expect that it may already be installed in system)

petr-nechaev, 2020-10-15:
Are you Ok with bumping CMake version to at 3.1 ?  If yes, I can implement the requested funcitonality, which I also need, and submit a PR.

M1cha, 2020-10-16:
which distros do have nanopb packages? IMO that's somewhere between useless and dangerous considering that you can change the struct layouts via defines.

petr-nechaev, 2020-10-16:
I think this github issue is not really about whether shared library shall be installed, or if nanopb shall be a distributed package or not.

It is really about partialy broken funcitonality ... which makes people loose time trying to understand why a simple damn thing that must work ... actually does not work.

We have 
```
install( FILES generator/proto/_utils.py
         DESTINATION ${PYTHON_INSTDIR}/proto/ )
```
This installs the generator into system Python modules. So the generator resides in /usr/local or /usr or whatever user installed it into.

BUT using `nanopb_generate_cpp()` function in CMake with the installed copy is not possible at all.   So nanopb must be used as a git sumbodule in-tree.

IMHO there are two good ways to fix this:
1. Remove cmake installation commands.  Make it impossible to run `make install`.  And say that nanopb must be used as a submodule in-tree only.    Makes a lot of sense actually.
2. Fix installation commands that the installed copy is fully-functional and does not require another in-tree clone.

petr-nechaev, 2020-10-16:
This issue is especially valid for complicated firmware build procedures like OpenEmbedded / Yocto.

petr-nechaev, 2020-10-16:
BTW  https://repology.org/project/nanopb/versions
https://packages.ubuntu.com/source/focal/nanopb

PetteriAimonen, 2023-11-11:
Ok, I have come to realize that this way of usage is rather common and should be fixed. #911 is another issue about this.
CMake is a bit foreign territory for me.