nanopb/nanopb #482: Remove deprecated "--nanopb_opt" flag from the cmake generator function

CoZZmOnAvT, 2020-01-20:
### Steps to reproduce the issue
1. mkdir -p build && cd build
2. cmake ../examples/cmake_simple
3. make

### What happens?
**_Error:_** Unknown flag: --nanopb_opt

### What should happen?
Generate C code for specified .proto file.

### How to Fix
Just remove deprecated flag form the **extra/FindNanopb.cmake**, line 259.

PetteriAimonen, 2020-01-21:
AFAIK that flag is not deprecated, but it is new as of protoc 3.6.

It was added to FindNanopb.cmake in pull request #454, to work around problems in passing options through the complicated escaping needed in `--nanopb_out=opts:outdir` format.

I'm open to anything regarding this, such as adding version detection or changing it to call nanopb_generator.py directly. But with my lacking CMake skills I'll need a pull request for it.

marekzukal, 2020-04-08:
#515 might solve it. It will allow to use the older way on non-Windows systems. Can someone with Windows prove that I did not mess anything up?

PetteriAimonen, 2020-09-21:
I think this should be solved now.

closed 2020-09-21 by PetteriAimonen